espressif/esp-at #255: Confuse of AT+RFPOWER? command

PowerfulCat, 2019-11-01:
Environment
- idf 3.3
- at 1.2.0
AT+RFPOWER?
+RFPOWER:-4,7,2,2

The wifi_power value read through this command is the dBm value, while the remaining three values are enumerate values. Is this the original intention of the design?

xcguang, 2019-11-01:
We have fixed it in the latest master branch, which will be released in the near two days.

PowerfulCat, 2019-11-01:
:)

xcguang, 2019-11-01:
the unit of the first value is dBm; the last three ones are power level; this is caused by different module APIs of the esp-idf.

Parameters:
- **\<wifi_power>**: range [40, 82], the unit is 0.25dBm, for example, if the value is 78, then RF max power is 78*0.25 dBm=19.5dBm
- **\<ble_adv_power>**: RF TX Power of BLE advertising, range: [0, 7]
    - 0:7dBm
    - 1:4dBm
    - 2:1dBm
    - 3:-2 dBm
    - 4:-5 dBm
    - 5:-8 dBm
    - 6:-11 dBm
    - 7:-14 dBm
- **\<ble_scan_power>**: RF TX Power of BLE scanning, range:  [0, 7], the same as **\<ble_adv_power>**
- **\<ble_conn_power>**: RF TX Power of BLE connecting, range:  [0, 7], the same as **\<ble_adv_power>**

xcguang, 2019-11-01:
+RFPOWER:-4,7,2,2    
`-4` is incorrect, which we fixed is.

xcguang, 2019-11-06:
We have released it, please try again.

closed 2020-01-09 by xcguang