espressif/esp-at #256: Potential problem of AT+CWLAP command

PowerfulCat, 2019-11-01:
+CWLAP:(3,"POT",-AL00a",-53,"80:56:xx:xx:xx:xx",1)
When set the ssid to 
POT",-AL00a
It will result a unpredictable behavior of string spliting.

marrakchino, 2019-11-19:
Do you mean when the SSID contains `"`?

PowerfulCat, 2019-11-20:
Yes, this command may be resolution fail if double quotation marks do not use escape characters, or make a complex  resolution coding.

PowerfulCat, 2019-11-20:
In current version, we have to resolution this response string from end to begin.

marrakchino, 2019-11-20:
I think it's already handled in the code
https://github.com/espressif/esp-at/blob/95a5aa9f5e4b954cda4bf3741bc0c86c1b088c80/examples/at_http_webserver/components/atparse/esp_parse_at.c#L286-L288
The end of the string is determined by the location of `,`, so every other character in between is considered (`"` makes no exception).

PowerfulCat, 2019-11-21:
There is no problem in AT part, this problem may be ignored in some analytical procedures.
+CWLAP:(3,"POT-",AL00a",-24,"12:34:56:78:9a:bc",6)␍␊
the ssid like this:
POT",-AL00a

I have to write a long code like this.
```C++
String current;
    tx(ASK_CWLAP);
    while((current = readLine()).length() == 0){
        delay(1);
    }
    auto lastIndexOf = [](const char * line, char chr, size_t times){
        for (int i = strlen(line); i-- > 0; ){
            if (line[i] != chr){
                continue;
            }
            if (--times == 0){
                return i;
            }
        }
        return -1;
    };

    while (current.indexOf(GET_CWLAP) != -1){
        //char    line[] = "+CWLAP:(3,\"POT\",-AL00a\",-53,\"12:34:56:78:9a:bc\",1)";
        char *  line = (char *)current.c_str();
        char *  p = nullptr;
        char *  t;
        int32_t mac[6]; //need 32bit when use sscanf
        int32_t ecn;
        int32_t rssi;
        int32_t channel;

        p = line + lastIndexOf(line, ',', 3);

        sscanf(line, GET_CWLAP "(%d,", & ecn);
        sscanf(p + 1, "%d,\"%x:%x:%x:%x:%x:%x\",%d", 
            & rssi,
            & mac[0],
            & mac[1],
            & mac[2],
            & mac[3],
            & mac[4],
            & mac[5],
            & channel
        );

        //replace '\"' to '\0'
        p[0] = ',';
        p[-1] = '\0';

        WifiLinkInfo info;
        info.ecn = ecn;
        info.ssid = strchr(line, ',') + 2; //skip ',' '\"'
        info.rssi = rssi;
        copy<uint8_t, int32_t>(info.bssid, mac, 6);
        info.channel = channel;
        call(info);
        current = readLine();
    }
```

xcguang, 2020-03-15:
#290

closed 2021-04-14 by FayeY