espressif/esp-at #259: Why ESP32 doesn't support AT command through SPI?

zslzxm, 2019-11-05:
Why ESP32 doesn't support AT command through SPI?
Is there have some hardware issue?
I have found the following code in this repo:
#if IDF_TARGET_ESP8266
source "$ESP_AT_PROJECT_PATH/main/interface/hspi/Kconfig"
#endif

Why only ESP8266 support?
Can I port this hspi to ESP32 too?
Thanks!

xcguang, 2019-11-05:
We have developed ESP32 SPI AT, but there is something wrong with it based on current driver, so we removed it. If you want to use MCU SPI, you can try ESP32 SDIO AT, MCU is transmit data through SPI. The demo for MCU is https://github.com/espressif/esp-at/tree/master/examples/at_sdspi_host, you just need to implement the APIs in  https://github.com/espressif/esp-at/blob/master/examples/at_sdspi_host/components/platform/esp32/include/port.h

zslzxm, 2019-11-05:
Hi, thank you for quickly reply.
My project use STM32 as the spi master, and ESP32 as spi slave.
I have noticed the at_sdspi_host example.
To use this example, we must connect STM32(SPI) to ESP32(SDIO), right?
But we have connect STM32 spi to ESP32 vspi interface(IO5, IO18, IO19...) already.
Does there have any other way to solve the issue, or we must re-design the hardware?
Thanks.

xcguang, 2019-11-05:
Hi,
SDIO pin cannot be changed. If porting spi to ESP32, maybe you need modify much ESP32 SPI driver. In the future, we will re-support it. But, currently, it seems  re-design the hardware is the better solution.  BTW, can you share your project schedule?

zslzxm, 2019-11-05:
Hi, wo need to deliver our products in 2 months later.
Do you have the plan when to support spi ?
We can  use uart at temporarily, but we want to support spi at ASAP, since the performance is better.

hwnBEAST, 2019-11-27:
> We have developed ESP32 SPI AT, but there is something wrong with it based on current driver, so we removed it. If you want to use MCU SPI, you can try ESP32 SDIO AT, MCU is transmit data through SPI. The demo for MCU is https://github.com/espressif/esp-at/tree/master/examples/at_sdspi_host, you just need to implement the APIs on MCU side in https://github.com/espressif/esp-at/blob/master/examples/at_sdspi_host/components/platform/esp32/include/port.h

Hello,
I have been developing AT stack on ESP32 over SPI to STM32 with the older version of HSPI. Can you tell me more about the issues with SPI AT?

xcguang, 2020-01-09:
@hwnBEAST If MCU want to send huge data, the data may be overwritten if ESP32 slave does not fetch in time.

tsvehagen, 2020-04-20:
@xcguang Is there any ongoing work on this issue?

xcguang, 2020-04-21:
Hi @tsvehagen ,
Sorry, now we are working on ESP8266 RTOS SDIO AT, and we will start ESP32 SPI AT after the end.
BTW, if your MCU only support SPI, you can use ESP32 SDIO AT,  there is a demo about it https://github.com/espressif/esp-at/tree/master/examples/at_sdspi_host

tsvehagen, 2020-04-22:
@xcguang Appreciate the answer, thanks. Yea the SDIO option would be nice but on the HiFive1 board (https://www.sifive.com/boards/hifive1-rev-b) not all the SDIO pins are not connected unfortunately.

MichaelDu9226, 2020-08-18:
@xcguang Is there any ongoing work on this issue?
In our usage, there is no need to be particularly fast or huge data, and we have added an IRQ pin.
I hope this feature will be released soon.

derSteFfi, 2020-10-29:
Would also look forward to AT via SPI since the hifive1_revb does not wire up all SDIO pins. Factory flashed firmware on the esp32 also uses HSPI-AT module so they got it to work somehow.

derSteFfi, 2021-02-04:
any updates on this? It would be really helpful to be able to at least reproduce the esp-at version that is running on Hifive1_revB boards.

takeyoshi, 2021-12-08:
Frankly speaking, all available documentation is either out of sync, outdated or just plainly wrong. There seems to be an "old" AT over SPI protocol and a new one. But neither the workflow is in-deeply explained nor the data_info or slave_info is (byte-order missing, details about slave-state missing), no explanation how the handshake-pin must be used, logic-analyzer images are all outdated (no dummy or address bytes as explained in the newer protocol). 

SDIO is gone in esp-at "menuconfig" too. 
Frankly speaking, I am starting to look for wifi-alternatives to the ESP (realtek, LSR, microchip, anything that actually works).

derSteFfi, 2022-01-13:
For anyone stumbling onto this: it seems like v2.2.0.0_esp32 is able to handle AT commands via SPI, I did not get around to test it yet, though.