espressif/esp-at #264: I can not compiling success when i use the hspi transmission mode.

PowerfulCat, 2019-11-12:
esp-at-1.2.0.0x/main/interface/hspi/at_hspi_task.c:488:109: error: 'module_name' undeclared (first use in this function)
 (char*)version),"Bin version:%s(%s)\r\n", CONFIG_ESP_AT_FW_VERSION, module_name

I remember the module_name has default value(WROOM-32), but it's still go wrong when i assign a new value by menuconfig.

ESP-Coco, 2019-11-13:
@PowerfulCat Hi  你试一下把 at_hspi_task.c 里面的第 484行到第491行删除，这部分在更新 AT 的时候没有更新 module_name 造成的。不过这部分不是必须项，可以直接删除。
不过我看你是要使用 ESP32 的 SPI，这部分我们的建议是你更换成 ESP32 的 SDIO（参考这个文档： https://github.com/espressif/esp-at/tree/master/examples/at_sdspi_host#%E7%AE%80%E4%BB%8B），对于MCU来说依然可以使用 SPI。
因为 ESP32 的 SPI 有一些硬伤无法解决：1. SPI clock 只支持到 10M 2. 在每次发完数据之后 MCU 需要 delay 一下，因为 handshake线被复用太多地方，导致 ESP32 的 SPI 挂载接收buffer的时间可能慢于 master 发送的速度，这块要想解决的话需要再加一根管脚，另外我们上层还进行了封装，导致实际速率还不如 3M 的 UART。

temp4gh, 2019-12-02:
esp32-at-2.0.0.0_esp8266/main/interface/hspi/at_hspi_task.c:41:30: fatal error: driver/spi_slave.h: No such file or directory
compilation terminated.
I tried to change the at interface from uart to hspi in menuconfig and got the above error. I wonder if the function of AT over hspi is only avalible on esp32?Thanx!

closed 2019-12-09 by PowerfulCat