espressif/esp-at #266: ESP32-AT V2.0 version use "AT+RFPOWER=0,0,0,0" reply error

zjykymt, 2019-11-13:
Hi,

When I upgraded to version 2.0, ESP32-WROOM-32D returned "ERROR" after sending "AT+RFPOWER=0,0,0,0".
- V1.2
 >AT+GMR
 AT version:1.2.0.0(6fe4345 - Jun 27 2019 05:39:23) 
 SDK version:v3.2-127-gddbce78 
 compile time:Jun 27 2019 12:08:01 
 Bin version:1.2.0(WROOM-32) 
 OK
 AT+RFPOWER=0,0,0,0
 OK

- V2.0
 >AT+GMR
 AT version:2.0.0.0(e44648c - Nov  4 2019 02:18:57) 
 SDK version:v3.3 
 compile time(7cac7ad):Nov  4 2019 06:11:40 
 Bin version:2.0.0(WROOM-32) 
 OK
 AT+RFPOWER=0,0,0,0
 ERROR

xcguang, 2019-11-13:
```
- **\<wifi_power>**: range [40, 82], the unit is 0.25dBm, for example, if the value is 78, then RF max power is 78*0.25 dBm=19.5dBm
- **\<ble_adv_power>**: RF TX Power of BLE advertising, range: [0, 7]
    - 0:7dBm
    - 1:4dBm
    - 2:1dBm
    - 3:-2 dBm
    - 4:-5 dBm
    - 5:-8 dBm
    - 6:-11 dBm
    - 7:-14 dBm
- **\<ble_scan_power>**: RF TX Power of BLE scanning, range:  [0, 7], the same as **\<ble_adv_power>**
- **\<ble_conn_power>**: RF TX Power of BLE connecting, range:  [0, 7], the same as **\<ble_adv_power>**
```

zjykymt, 2019-11-13:
Ok, i got it. Please add this change to the description of release if you can.

Thanks!

reshma26p, 2019-11-13:
i got following respons when esp32 bluetooth device connected to PC...Please give meaning of respons
_bt_gap_cb - event = 4

xiewenxiang, 2020-01-09:
Please provide the full log.

closed 2020-03-15 by xcguang