espressif/esp-at #267: BT command error

reshma26p, 2019-11-13:
by connecting esp BT to PC manually gives response as
at_bt_gap_cb - event = 1

at_bt_gap_cb - event = 4
by connecting esp BT to PC by using AT command 
 AT+BTSPPCONN=0,0,"24:0a:c4:09:34:23"
gives response
at_bt_spp_cb - event = 8
ERROR

please give me response urgently

xiewenxiang, 2019-11-14:
Please firstly discovery the peripheral devices and check if the remote device exists.

reshma26p, 2019-11-14:
yes device  discovery available...but while connecting using AT command response as
at_bt_spp_cb - event = 8
ERROR
and also checking connected device using command AT+BTSPPCONN? it gives response as +BTSPPCONN:-1 even if device is connected

reshma26p, 2019-11-14:
please give response its to much urgent for me i am trying for it from 7-8 days

reshma26p, 2019-11-18:
@xiewenxiang please help me i am using esp32 at v2.0.0.0_esp32 and IDF v3.3
and following instructions given in get started ...is it possible to issue in code

xiewenxiang, 2019-11-18:
@reshma26p 

Do you initial the  spp profile?

`AT+BTSPPINIT=1` or `AT+BTSPPINIT=2`

https://github.com/espressif/esp-at/blob/master/docs/ESP32_AT_Classic_Bluetooth.md

reshma26p, 2019-11-18:
@xiewenxiang 
yes i initiate spp profile using AT+BTSPPINIT=2  and getting response ok

xiewenxiang, 2019-11-18:
If you want to connect to the device, please try `AT+BTSPPINIT=1`

reshma26p, 2019-11-18:
yes i am also try this..device get paired but not connect 

AT+BTSPPCONN=0,0,"24:da:9b:10:0f:36"
after some time gives respons as 
ERROR

xiewenxiang, 2019-11-18:
Can you try to initiate the connection request with PC？

closed 2019-11-28 by reshma26p