espressif/esp-at #268: at_customize.csv differs between v1.2.0.0 (esp32) and v2.0.0.0_esp32. Problems if OTA update?

carlessole, 2019-11-13:
If passing from [v1.2.0.0](https://github.com/espressif/esp-at/releases/tag/v1.2.0.0) to [v2.0.0.0_esp32](https://github.com/espressif/esp-at/releases/tag/v2.0.0.0_esp32) via OTA update, I assume that just the OTA partition will be updated. 
Then all the content on other partitions will remain untouched, is it true (all the rows in purple)?
[image]

Then, between this two versions, at_customize is not the same:
[image]

That could cause any malfunction when updating AT version via OTA update?
I assume that everything will work fine but.. if I want to use some of the MQTT partition like "mqtt_cert", then there is no way to do that no?

Thanks!

xcguang, 2019-11-14:
Yes, maybe the project should add this feature.

reshma26p, 2019-11-14:
@xcguang  please give me response to #267

carlessole, 2019-11-14:
Thanks @xcguang 
But... I don't understand what are you referring with this sentence:
> Yes, maybe the project should add this feature.

Could you explain it with more detail?

So my questions are:

1. Then all the content on other partitions will remain untouched, is it true (all the rows in purple)?
2. That could cause any malfunction when updating AT version via OTA update?
3. If I want to use some of the MQTT partition like "mqtt_cert", then there is no way to do that no? How ti has to be done? The code of OTA partition should be able to write to at_customize partition?

Thank you very much

xcguang, 2019-11-14:
`Then all the content on other partitions will remain untouched, is it true (all the rows in purple)?`
-- Currently, it is.

`That could cause any malfunction when updating AT version via OTA update?`
-- If you do not use the new feature, it should work well.

`If I want to use some of the MQTT partition like "mqtt_cert", then there is no way to do that no? How ti has to be done? The code of OTA partition should be able to write to at_customize partition?`
-- Yes,  in my opinion, OTA at_customize or mqtt cert should be supported. But the user should make sure the OTA is correctly completed. If fails, user should re-ota it.

closed 2020-01-09 by xcguang