espressif/esp-at #269: Example at_http_webserver always rebooting with esp-idf v3.3

wildwing1970, 2019-11-13:
Hi Guys,

i am a newbie to esp-32 and tried the at_http_webserver example, but it just crashed!

After some investigation i found out what was missing ... here is the code

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "nvs_flash.h"
#include "esp_log.h"
#include "esp_system.h"
#include "esp_partition.h"

#include "esp_event_loop.h"
#include "esp_wifi.h"

#include "esp_at.h"
//#include "at_interface.h"

#include "espfs.h"
#include "httpd.h"
#include "httpdespfs.h"
#include "cgiat.h"
#include "atparse.h"

static const char* TAG = "HTTPAT";

HttpdBuiltInUrl builtInUrls[] = {
    {"/", cgiRedirect, "/login.html"},
    {"/routeinfo", cgiRouteInfo, NULL},
    {"/showapinfo", cgiShowApInfo, NULL},
    {"/configapinfo", cgiConfigApInfo, NULL},
    {"/connap", cgiConnAp, NULL},
    {"/getramsize", getRamSize, NULL},
    {"/ping", cgiping, NULL},
    {"/rst", cgirst, NULL},
    {"/restore", cgirestore, NULL},
    {"/loginpwd", cgiloginpwd, NULL},
    {"/changeloginpwd", cgiChangeLoginPwd, NULL},
    {"/closeap", cgiCloseAp, NULL},
    {"/setap", cgiSetAp, NULL},
    {"/scanap", cgiScanAp, NULL},
    {"/setdhcp", cgiSetDhcp, NULL},
    {"/showdhcpinfo", cgiShowDhcp, NULL},
    {"/setstaipinfo", cgiSetStaIpInfo, NULL},
    {"/setapipinfo", cgiSetApIpInfo, NULL},
    {"/setstamac", cgiSetStaMac, NULL},
    {"/setapmac", cgiSetApMac, NULL},
    {"/connip", cgiConnIp, NULL},
    {"/closeip", cgiCloseIp, NULL},
    {"/getipstatus", cgiGetIpStatus, NULL},
    {"/softapdevice", getConnDev, NULL},

    {"/bleset", cgiBleSet, NULL},
    {"/blescan", cgiBleScan, NULL},
    {"/getbleset", cgiGetBleSet, NULL},
    {"/bleconnect", cgiBleConn, NULL},
    {"/getbleconnect", cgiGetBleConn, NULL},
    {"/blecloseconnect", cgiBleCloseConn, NULL},
    {"/bleadvset", cgiBleAdvSet, NULL},
    {"/getbleadvset", cgiGetBleAdvSet, NULL},
    {"/bleadvstart", cgiBleAdvStart, NULL},
    {"/bleadvstop", cgiBleAdvStop, NULL},
    {"*", cgiEspFsHook, NULL}, //Catch-all cgi function for the filesystem
    {NULL, NULL, NULL}
};

static esp_err_t at_wifi_event_handler(void *ctx, system_event_t *event)
{
    esp_err_t ret = esp_at_wifi_event_handler(ctx, event);

    return ret;
}

static void initialise_wifi(void)
{
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();

    esp_event_loop_init(at_wifi_event_handler, NULL);
    esp_wifi_init(&cfg);
    esp_wifi_set_storage(WIFI_STORAGE_RAM);
    esp_wifi_start();
}

void app_main()
{
    nvs_flash_init();
    tcpip_adapter_init();

    initialise_wifi();
    esp_at_init_parse();

    espFsInit();

    httpdInit(builtInUrls, 80);
}

ESP-Coco, 2019-11-14:
Hi @wildwing1970 Thank you very much :) . The method you provided is correct. Indeed, all the samples were not re-adapted during the process of updating AT. I will update it as soon as possible.

xcguang, 2019-12-10:
Has updated.

closed 2019-12-10 by xcguang