espressif/esp-at #270: [ESP32 AT UART]: Any size limitation in UART?

zslzxm, 2019-11-14:
Hi,
The data path is: STM32 <- uart at ->ESP32<- wifi ->Mobile APP.
After esp32 connect to the tcp server in APP, the APP sends 2915 bytes to esp32.
Now the issue happen, some time STM32 receive the correct data, some time STM32 receive few bytes.
[The normal case]:
+IPD,2915:{"datalength":"00000b3c","ver":"1.0.0"}{"cmd":"upgrade","crc":"732786823","data":"REVTQ19GTF9TSEFSRUQpJiYodmQtPmZsYWdzJlZFQ0RFU0NfRkxfTk9OU0hBUkVEKSkAAHN2ZABlc3BfaW50cl9kaXNhYmxlAAAAAGVzcF9pbnRyX2ZyZWUAAABpc192ZWN0X2Rlc2NfdXNhYmxlAGZpbmRfZGVzY19mb3Jfc291cmNlAAAAAAEAAAAAAAAAAQAAAAEAAAABAAAAAAAAAAEAAAABAAAAAQAAAAAAAAAAAAAAAAAAAAEAAAAAAAAAAAAAAAAAAAABAAAAAAAAAAEAAAAAAAAAAQAAAAAAAAABAAAAAQAAAAEAAAACAAAAAQAAAAEAAAABAAAAAgAAAAIAAAACAAAAAQAAAAAAAAABAAAAAQAAAAEAAAAAAAAAAAAAAAAAAAABAAAAAQAAAAAAAAAAAAAAAwAAAAIAAAACAAAAAgAAAAEAAAAAAAAAAAAAAAAAAAABAAAAAAAAAAAAAAAAAAAABwAAAAAAAAABAAAAAQAAAAMAAAACAAAAAgAAAAIAAAAFAAAAAgAAAAIAAAACAAAAAQAAAAAAAAAAAAAAAAAAAAEAAAAAAAAAAAAAAAAAAAACAAAAAAAAAAAAAAAAAAAAAgAAAAAAAAAAAAAAAAAAAAIAAAAAAAAAAAAAAAAAAAADAAAAAQAAAAEAAAAAAAAAAwAAAAAAAAAAAAAAAAAAAAQAAAAAAAAAAQAAAAAAAAAEAAAAAAAAAAEAAAABAAAABQAAAAAAAAAAAAAAAQAAAAMAAAAAAAAAAQAAAAEAAAAEAAAAAQAAAAAAAAAAAAAAAwAAAAIAAAACAAAAAgAAAAQAAAABAAAAAQAAAAEAAAAFAAAAAAAAAAEAAAABAAAAc3lzdGVtX2FwaQAAG1swOzMxbUUgKCVkKSAlczogQmFzZSBNQUMgYWRkcmVzcyBpcyBOVUxMG1swbQoAG1swOzMybUkgKCVkKSAlczogQmFzZSBNQUMgYWRkcmVzcyBpcyBub3Qgc2V0LCByZWFkIGRlZmF1bHQgYmFzZSBNQUMgYWRkcmVzcyBmcm9tIEJMSzAgb2YgRUZVU0UbWzBtCgAAAAAbWzA7MzFtRSAoJWQpICVzOiBCYXNlIE1BQyBhZGRyZXNzIGZyb20gQkxLMyBvZiBFRlVTRSB2ZXJzaW9uIGVycm9yLCB2ZXJzaW9uID0gJWQbWzBtCgAAG1swOzMxbUUgKCVkKSAlczogQmFzZSBNQUMgYWRkcmVzcyBmcm9tIEJMSzMgb2YgRUZVU0UgQ1JDIGVycm9yLCBlZnVzZV9jcmMgPSAweCUwMng7IGNhbGNfY3JjID0gMHglMDJ4G1swbQoAG1swOzMxbUUgKCVkKSAlczogQmFzZSBNQUMgYWRkcmVzcyBmcm9tIEJMSzAgb2YgRUZVU0UgQ1JDIGVycm9yLCBlZnVzZV9jcmMgPSAweCUwMng7IGNhbGNfY3JjID0gMHglMDJ4G1swbQoAG1swOzMxbUUgKCVkKSAlczogbWFjIGFkZHJlc3MgcGFyYW0gaXMgTlVMTBtbMG0KAAAAABtbMDszMW1FICglZCkgJXM6IG1hYyB0eXBlIGlzIGluY29ycmVjdBtbMG0KAAAAABtbMDszM21XICglZCkgJXM6IGluY29ycmVjdCBtYWMgdHlwZRtbMG0KAAAAbWFzdGVyAAB2NC4xLWRldi0yODEtZzk2Yjk2YWUyNC1kaXJ0eQAAAHRhc2tfd2R0AAAAABtbMDszMW1FICglZCkgJXM6IFRhc2sgd2F0Y2hkb2cgZ290IHRyaWdnZXJlZC4gVGhlIGZvbGxvd2luZyB0YXNrcyBkaWQgbm90IHJlc2V0IHRoZSB3YXRjaGRvZyBpbiB0aW1lOhtbMG0KABtbMDszMW1FICglZCkgJXM6ICAtICVzICglcykbWzBtCgAAABtbMDszMW1FICglZCkgJXM6ICVzG1swbQoAAAAbWzA7MzFtRSAoJWQpICVzOiBDUFUgJWQ6ICVzG1swbQoAAAAbWzA7MzFtRSAoJWQpICVzOiBBYm9ydGluZy4bWzBtCgAAAABlc3BfaW50cl9hbGxvYyhFVFNfVEcwX1dEVF9MRVZFTF9JTlRSX1NPVVJDRSwgMCwgdGFza193ZHRfaXNyLCBOVUxMLCAmdHdkdF9jb25maWctPmludHJfaGFuZGxlKQBDOi9Vc2Vycy9tYXJremhhbmcvZXNwL2VzcC1pZGYvY29tcG9uZW50cy9lc3AzMi90YXNrX3dkdC5jAABlc3BfaW50cl9mcmVlKHR3ZHRfY29uZmlnLT5pbnRyX2hhbmRsZSkAZXNwX3JlZ2lzdGVyX2ZyZWVydG9zX2lkbGVfaG9va19mb3JfY3B1KGlkbGVfaG9va19jYiwgaSkAAAAAZXNwX3Rhc2tfd2R0X2FkZAAAAABlc3BfdGFza193ZHRfaW5pdAAAAA0KQnJvd25vdXQgZGV0ZWN0b3Igd2FzIHRyaWdnZXJlZA0KDQoAAABydGNfaXNyX3JlZ2lzdGVyKHJ0Y19icm93bm91dF9pc3JfaGFuZGxlciwgTlVMTCwgUlRDX0NOVExfQlJPV05fT1VUX0lOVF9FTkFfTSkAAEM6L1VzZXJzL21hcmt6aGFuZy9lc3AvZXNwLWlkZi9jb21wb25lbnRzL2VzcDMyL2Jyb3dub3V0LmMAAGVzcF9icm93bm91dF9pbmk=","filename":"MS_WIFI_V1.0.1.bin","fwtype":"5","seqno":"1","totalpackets":"326","totalsize":"667200"}

[The abnormal case]:
+IPD,2872:{"datalength":"00000b3c","ver":"1.0.0"}{"cmd":"upgrade","crc":"314500239","data":"6QYCEGwSCEDuAAAAAAAAAAAAAAAAAAABIABAPwC2AQAyVM2rAAAAAAAAAAAAAAAAY2UxY2RmOC1kaXJ0eQAAAAAAAAAAAAAAAAAAAAAAAABkb3Rvbl9lc3AzMgAAAAAAAAAAAAAAAAAAAAAAAAAAADE0OjE2OjQwAAAAAAAAAABPY3QgMzEgMjAxOQAAAAAAdjQuMS1kZXYtMjgxLWc5NmI5NmFlMjQtZGlydHkAAACVfFCSUriR0eeb0W9cprMmHrJqPm45dRHgmeO5y6A8awAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAIAAAACgAAAAwAAAAOBlc3BfdGFza193ZHRfaW5pdChDT05GSUdfRVNQX1RBU0tfV0RUX1RJTUVPVVRfUywgZmFsc2UpAEM6L1VzZXJzL21hcmt6aGFuZy9lc3AvZXNwLWlkZi9jb21wb25lbnRzL2VzcDMyL2NwdV9zdGFydC5jAGVzcF90YXNrX3dkdF9hZGQoaWRsZV8wKQAAAABlc3BfdGFza193ZHRfYWRkKGlkbGVfMSkAAAAAL2Rldi91YXJ0LzAAZXJyID09IEVTUF9PSyAmJiAiRmFpbGVkIHRvIGluaXQgcHRocmVhZCBtb2R1bGUhIgAAAGZsYXNoX3JldCA9PSBFU1BfT0sAY3B1X3N0YXJ0AAAAG1swOzMybUkgKCVkKSAlczogQ2hpcCBSZXZpc2lvbjogJWQbWzBtCgAAAAAbWzA7MzNtVyAoJWQpICVzOiBDaGlwIHJldmlzaW9uIGlzIGhpZ2hlciB0aGFuIHRoZSBvbmUgY29uZmlndXJlZCBpbiBtZW51Y29uZmlnLiBTdWdnZXN0IHRvIHVwZ3JhZGUgaXQuG1swbQoAAAAAG1swOzMxbUUgKCVkKSAlczogRVNQLUlERiBjYW4ndCBzdXBwb3J0IHRoaXMgY2hpcCByZXZpc2lvbi4gTW9kaWZ5IG1pbmltdW0gc3VwcG9ydGVkIHJldmlzaW9uIGluIG1lbnVjb25maWcbWzBtCgAAAABtYWluAAAAAHJlcyA9PSBwZFRSVUUAAAAbWzA7MzJtSSAoJWQpICVzOiBTdGFydGluZyBzY2hlZHVsZXIgb24gUFJPIENQVS4bWzBtCgAAABtbMDszMm1JICglZCkgJXM6IFN0YXJ0aW5nIHNjaGVkdWxlciBvbiBBUFAgQ1BVLhtbMG0KAAAAG1swOzMybUkgKCVkKSAlczogQXBwIGNwdSB1cC4bWzBtCgAAG1swOzMybUkgKCVkKSAlczogUHJvIGNwdSB1cC4bWzBtCgAAG1swOzMybUkgKCVkKSAlczogQXBwbGljYXRpb24gaW5mb3JtYXRpb246G1swbQoAG1swOzMybUkgKCVkKSAlczogUHJvamVjdCBuYW1lOiAgICAgJXMbWzBtCgAbWzA7MzJtSSAoJWQpICVzOiBBcHAgdmVyc2lvbjogICAgICAlcxtbMG0KABtbMDszMm1JICglZCkgJXM6IENvbXBpbGUgdGltZTogICAgICVzICVzG1swbQoAABtbMDszMm1JICglZCkgJXM6IEVMRiBmaWxlIFNIQTI1NjogICVzLi4uG1swbQoAABtbMDszMm1JICglZCkgJXM6IEVTUC1JREY6ICAgICAgICAgICVzG1swbQoAG1swOzMxbUUgKCVkKSAlczogUnVubmluZyBvbiBzaW5nbGUgY29yZSBjaGlwLCBidXQgYXBwbGljYXRpb24gaXMgYnVpbHQgd2l0aCBkdWFsIGNvcmUgc3VwcG9ydC4bWzBtCgAAAAAbWzA7MzFtRSAoJWQpICVzOiBQbGVhc2UgZW5hYmxlIENPTkZJR19GUkVFUlRPU19VTklDT1JFIG9wdGlvbiBpbiBtZW51Y29uZmlnLhtbMG0KAAAbWzA7MzJtSSAoJWQpICVzOiBTdGFydGluZyBhcHAgY3B1LCBlbnRyeSBwb2ludCBpcyAlcBtbMG0KAAAbWzA7MzJtSSAoJWQpICVzOiBQcm8gY3B1IHN0YXJ0IHVzZXIgY29kZRtbMG0KAABtYWluX3Rhc2sAAABzdGFydF9jcHUwX2RlZmF1bHQAAGNvcmVfaWQ8cG9ydE5VTV9QUk9DRVNTT1JTAABDOi9Vc2Vycy9tYXJremhhbmcvZXNwL2VzcC1pZGYvY29tcG9uZW50cy9lc3AzMi9jcm9zc2NvcmVfaW50LmMAZXJyID09IEVTUF9PSwAAAGVzcF9jcm9zc2NvcmVfaW50X3NlbmQAAGVzcF9jcm9zc2NvcmVfaW50X2luaXQAAEM6L1VzZXJzL21hcmt6aGFuZy9lc3AvZXNwLWlkZi9jb21wb25lbnRzL2VzcDMyL2Rwb3J0X2FjY2Vzcy5jAABkcG9ydAAAAGVzcF9kcG9ydF9hY2Nlc3NfaW50X2luaXQAAABlc3BfZHBvcnRfYWNjZXNzX3N0YWxsX290aGVyX2NwdV9lbmQAAAAAc3ZkICE9IE5VTEwAQzovVXNlcnMvbWFya3poYW5nL2VzcC9lc3AtaWRmL2NvbXBvbmVudHMvZXNwMzIvaW50cl9hbGxvYy5jAAAAACEoKHZkLT5mbGFncyZWRUM=","filename":"MS_WIFI_V1.0.1.bin","fwtype":"5","seqno":"0"

The abnormal case receive few bytes.
1. Why this case happen?
2. It seems the esp32 will not send the remaining bytes after the abnomal case. How to avoid this?

Thanks!

zslzxm, 2019-11-14:
[The abnormal case]:
+IPD,2872:{"datalength":"00000b3c","ver":"1.0.0"}{"cmd":"upgrade","crc":"314500239","data":"6QYCEGwSCEDuAAAAAAAAAAAAAAAAAAABIABAPwC2AQAyVM2rAAAAAAAAAAAAAAAAY2UxY2RmOC1kaXJ0eQAAAAAAAAAAAAAAAAAAAAAAAABkb3Rvbl9lc3AzMgAAAAAAAAAAAAAAAAAAAAAAAAAAADE0OjE2OjQwAAAAAAAAAABPY3QgMzEgMjAxOQAAAAAAdjQuMS1kZXYtMjgxLWc5NmI5NmFlMjQtZGlydHkAAACVfFCSUriR0eeb0W9cprMmHrJqPm45dRHgmeO5y6A8awAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAIAAAACgAAAAwAAAAOBlc3BfdGFza193ZHRfaW5pdChDT05GSUdfRVNQX1RBU0tfV0RUX1RJTUVPVVRfUywgZmFsc2UpAEM6L1VzZXJzL21hcmt6aGFuZy9lc3AvZXNwLWlkZi9jb21wb25lbnRzL2VzcDMyL2NwdV9zdGFydC5jAGVzcF90YXNrX3dkdF9hZGQoaWRsZV8wKQAAAABlc3BfdGFza193ZHRfYWRkKGlkbGVfMSkAAAAAL2Rldi91YXJ0LzAAZXJyID09IEVTUF9PSyAmJiAiRmFpbGVkIHRvIGluaXQgcHRocmVhZCBtb2R1bGUhIgAAAGZsYXNoX3JldCA9PSBFU1BfT0sAY3B1X3N0YXJ0AAAAG1swOzMybUkgKCVkKSAlczogQ2hpcCBSZXZpc2lvbjogJWQbWzBtCgAAAAAbWzA7MzNtVyAoJWQpICVzOiBDaGlwIHJldmlzaW9uIGlzIGhpZ2hlciB0aGFuIHRoZSBvbmUgY29uZmlndXJlZCBpbiBtZW51Y29uZmlnLiBTdWdnZXN0IHRvIHVwZ3JhZGUgaXQuG1swbQoAAAAAG1swOzMxbUUgKCVkKSAlczogRVNQLUlERiBjYW4ndCBzdXBwb3J0IHRoaXMgY2hpcCByZXZpc2lvbi4gTW9kaWZ5IG1pbmltdW0gc3VwcG9ydGVkIHJldmlzaW9uIGluIG1lbnVjb25maWcbWzBtCgAAAABtYWluAAAAAHJlcyA9PSBwZFRSVUUAAAAbWzA7MzJtSSAoJWQpICVzOiBTdGFydGluZyBzY2hlZHVsZXIgb24gUFJPIENQVS4bWzBtCgAAABtbMDszMm1JICglZCkgJXM6IFN0YXJ0aW5nIHNjaGVkdWxlciBvbiBBUFAgQ1BVLhtbMG0KAAAAG1swOzMybUkgKCVkKSAlczogQXBwIGNwdSB1cC4bWzBtCgAAG1swOzMybUkgKCVkKSAlczogUHJvIGNwdSB1cC4bWzBtCgAAG1swOzMybUkgKCVkKSAlczogQXBwbGljYXRpb24gaW5mb3JtYXRpb246G1swbQoAG1swOzMybUkgKCVkKSAlczogUHJvamVjdCBuYW1lOiAgICAgJXMbWzBtCgAbWzA7MzJtSSAoJWQpICVzOiBBcHAgdmVyc2lvbjogICAgICAlcxtbMG0KABtbMDszMm1JICglZCkgJXM6IENvbXBpbGUgdGltZTogICAgICVzICVzG1swbQoAABtbMDszMm1JICglZCkgJXM6IEVMRiBmaWxlIFNIQTI1NjogICVzLi4uG1swbQoAABtbMDszMm1JICglZCkgJXM6IEVTUC1JREY6ICAgICAgICAgICVzG1swbQoAG1swOzMxbUUgKCVkKSAlczogUnVubmluZyBvbiBzaW5nbGUgY29yZSBjaGlwLCBidXQgYXBwbGljYXRpb24gaXMgYnVpbHQgd2l0aCBkdWFsIGNvcmUgc3VwcG9ydC4bWzBtCgAAAAAbWzA7MzFtRSAoJWQpICVzOiBQbGVhc2UgZW5hYmxlIENPTkZJR19GUkVFUlRPU19VTklDT1JFIG9wdGlvbiBpbiBtZW51Y29uZmlnLhtbMG0KAAAbWzA7MzJtSSAoJWQpICVzOiBTdGFydGluZyBhcHAgY3B1LCBlbnRyeSBwb2ludCBpcyAlcBtbMG0KAAAbWzA7MzJtSSAoJWQpICVzOiBQcm8gY3B1IHN0YXJ0IHVzZXIgY29kZRtbMG0KAABtYWluX3Rhc2sAAABzdGFydF9jcHUwX2RlZmF1bHQAAGNvcmVfaWQ8cG9ydE5VTV9QUk9DRVNTT1JTAABDOi9Vc2Vycy9tYXJremhhbmcvZXNwL2VzcC1pZGYvY29tcG9uZW50cy9lc3AzMi9jcm9zc2NvcmVfaW50LmMAZXJyID09IEVTUF9PSwAAAGVzcF9jcm9zc2NvcmVfaW50X3NlbmQAAGVzcF9jcm9zc2NvcmVfaW50X2luaXQAAEM6L1VzZXJzL21hcmt6aGFuZy9lc3AvZXNwLWlkZi9jb21wb25lbnRzL2VzcDMyL2Rwb3J0X2FjY2Vzcy5jAABkcG9ydAAAAGVzcF9kcG9ydF9hY2Nlc3NfaW50X2luaXQAAABlc3BfZHBvcnRfYWNjZXNzX3N0YWxsX290aGVyX2NwdV9lbmQAAAAAc3ZkICE9IE5VTEwAQzovVXNlcnMvbWFya3poYW5nL2VzcC9lc3AtaWRmL2NvbXBvbmVudHMvZXNwMzIvaW50cl9hbGxvYy5jAAAAACEoKHZkLT5mbGFncyZWRUM=","filename":"MS_WIFI_V1.0.1.bin","fwtype":"5","seqno":"0"

zslzxm, 2019-11-14:
....
[The abnormal case]
+IPD,2872:{"datalength":"00000b3c","ver":"1.0.0"}{"cmd":"upgrade","crc":"314500239","data":"6QYCEGwSCEDuAAAAAAAAAAAAAAAAAAABIABAPwC2AQAyVM2rAAAAAAAAAAAAAAAAY2UxY2RmOC1kaXJ0eQAAAAAAAAAAAAAAAAAAAAAAAABkb3Rvbl9lc3AzMgAAAAAAAAAAAAAAAAAAAAAAAAAAADE0OjE2OjQwAAAAAAAAAABPY3QgMzEgMjAxOQAAAAAAdjQuMS1kZXYtMjgxLWc5NmI5NmFlMjQtZGlydHkAAACVfFCSUriR0eeb0W9cprMmHrJqPm45dRHgmeO5y6A8awAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAIAAAACgAAAAwAAAAOBlc3BfdGFza193ZHRfaW5pdChDT05GSUdfRVNQX1RBU0tfV0RUX1RJTUVPVVRfUywgZmFsc2UpAEM6L1VzZXJzL21hcmt6aGFuZy9lc3AvZXNwLWlkZi9jb21wb25lbnRzL2VzcDMyL2NwdV9zdGFydC5jAGVzcF90YXNrX3dkdF9hZGQoaWRsZV8wKQAAAABlc3BfdGFza193ZHRfYWRkKGlkbGVfMSkAAAAAL2Rldi91YXJ0LzAAZXJyID09IEVTUF9PSyAmJiAiRmFpbGVkIHRvIGluaXQgcHRocmVhZCBtb2R1bGUhIgAAAGZsYXNoX3JldCA9PSBFU1BfT0sAY3B1X3N0YXJ0AAAAG1swOzMybUkgKCVkKSAlczogQ2hpcCBSZXZpc2lvbjogJWQbWzBtCgAAAAAbWzA7MzNtVyAoJWQpICVzOiBDaGlwIHJldmlzaW9uIGlzIGhpZ2hlciB0aGFuIHRoZSBvbmUgY29uZmlndXJlZCBpbiBtZW51Y29uZmlnLiBTdWdnZXN0IHRvIHVwZ3JhZGUgaXQuG1swbQoAAAAAG1swOzMxbUUgKCVkKSAlczogRVNQLUlERiBjYW4ndCBzdXBwb3J0IHRoaXMgY2hpcCByZXZpc2lvbi4gTW9kaWZ5IG1pbmltdW0gc3VwcG9ydGVkIHJldmlzaW9uIGluIG1lbnVjb25maWcbWzBtCgAAAABtYWluAAAAAHJlcyA9PSBwZFRSVUUAAAAbWzA7MzJtSSAoJWQpICVzOiBTdGFydGluZyBzY2hlZHVsZXIgb24gUFJPIENQVS4bWzBtCgAAABtbMDszMm1JICglZCkgJXM6IFN0YXJ0aW5nIHNjaGVkdWxlciBvbiBBUFAgQ1BVLhtbMG0KAAAAG1swOzMybUkgKCVkKSAlczogQXBwIGNwdSB1cC4bWzBtCgAAG1swOzMybUkgKCVkKSAlczogUHJvIGNwdSB1cC4bWzBtCgAAG1swOzMybUkgKCVkKSAlczogQXBwbGljYXRpb24gaW5mb3JtYXRpb246G1swbQoAG1swOzMybUkgKCVkKSAlczogUHJvamVjdCBuYW1lOiAgICAgJXMbWzBtCgAbWzA7MzJtSSAoJWQpICVzOiBBcHAgdmVyc2lvbjogICAgICAlcxtbMG0KABtbMDszMm1JICglZCkgJXM6IENvbXBpbGUgdGltZTogICAgICVzICVzG1swbQoAABtbMDszMm1JICglZCkgJXM6IEVMRiBmaWxlIFNIQTI1NjogICVzLi4uG1swbQoAABtbMDszMm1JICglZCkgJXM6IEVTUC1JREY6ICAgICAgICAgICVzG1swbQoAG1swOzMxbUUgKCVkKSAlczogUnVubmluZyBvbiBzaW5nbGUgY29yZSBjaGlwLCBidXQgYXBwbGljYXRpb24gaXMgYnVpbHQgd2l0aCBkdWFsIGNvcmUgc3VwcG9ydC4bWzBtCgAAAAAbWzA7MzFtRSAoJWQpICVzOiBQbGVhc2UgZW5hYmxlIENPTkZJR19GUkVFUlRPU19VTklDT1JFIG9wdGlvbiBpbiBtZW51Y29uZmlnLhtbMG0KAAAbWzA7MzJtSSAoJWQpICVzOiBTdGFydGluZyBhcHAgY3B1LCBlbnRyeSBwb2ludCBpcyAlcBtbMG0KAAAbWzA7MzJtSSAoJWQpICVzOiBQcm8gY3B1IHN0YXJ0IHVzZXIgY29kZRtbMG0KAABtYWluX3Rhc2sAAABzdGFydF9jcHUwX2RlZmF1bHQAAGNvcmVfaWQ8cG9ydE5VTV9QUk9DRVNTT1JTAABDOi9Vc2Vycy9tYXJremhhbmcvZXNwL2VzcC1pZGYvY29tcG9uZW50cy9lc3AzMi9jcm9zc2NvcmVfaW50LmMAZXJyID09IEVTUF9PSwAAAGVzcF9jcm9zc2NvcmVfaW50X3NlbmQAAGVzcF9jcm9zc2NvcmVfaW50X2luaXQAAEM6L1VzZXJzL21hcmt6aGFuZy9lc3AvZXNwLWlkZi9jb21wb25lbnRzL2VzcDMyL2Rwb3J0X2FjY2Vzcy5jAABkcG9ydAAAAGVzcF9kcG9ydF9hY2Nlc3NfaW50X2luaXQAAABlc3BfZHBvcnRfYWNjZXNzX3N0YWxsX290aGVyX2NwdV9lbmQAAAAAc3ZkICE9IE5VTEwAQzovVXNlcnMvbWFya3poYW5nL2VzcC9lc3AtaWRmL2NvbXBvbmVudHMvZXNwMzIvaW50cl9hbGxvYy5jAAAAACEoKHZkLT5mbGFncyZWRUM=","filename":"MS_WIFI_V1.0.1.bin","fwtype":"5","seqno":"0"

boyhuesd, 2019-11-14:
+IPD max length is about 4096 bytes if I remember correctly.
I think it will send in chunks, so you will need to accumulate it to get full data. For example, result is 2048 bytes, it may send +IPD,1024 then another +IPD,1024.
Hope it helps.

xcguang, 2020-01-09:
Or you can try `AT+CIPRECVMODE` to receive the length you specify.

xcguang, 2020-01-09:
If you have any question, please reopen it.

closed 2020-01-09 by xcguang