espressif/esp-at #271: [BLE_CLIENT] Custom AT+BLEGATTCWR

boyhuesd, 2019-11-14:
Hi guys,

I'm facing an issue with AT+BLEGATTCWR command. The problem is for some devices it does not write to the correct handle of the BLE characteristic. I'm not sure what is the main cause of this, but for some devices, the corrected handle for a write/write no response UUID is characteristic handle+1 (Please correct me if i'm wrong).
If using the SDK API, I think I can control it, but since we are using AT command, I'm not sure if there is any way to resolve this issue.
So I'm thinking about below. Please help me if you think these one are doable or any thing else.
1) Any current AT command thats support update a characteristic handle.
2) If there is no AT command like in 1), I think I will need to add a custom AT command for BLE write with esp32-at source. I think the right API for GATTCWR is:

`
/**
 * @brief           This function is called to write characteristic value.
 *
 * @param[in]       gattc_if: Gatt client access interface.
 * @param[in]       conn_id : connection ID.
 * @param[in]       handle : characteristic handle to write.
 * @param[in]       value_len: length of the value to be written.
 * @param[in]       value : the value to be written.
 * @param[in]       write_type : the type of attribute write operation.
 * @param[in]       auth_req : authentication request.
 *
 * @return
 *                  - ESP_OK: success
 *                  - other: failed
 *
 */
esp_err_t esp_ble_gattc_write_char( esp_gatt_if_t gattc_if,
                                    uint16_t conn_id,
                                    uint16_t handle,
                                    uint16_t value_len,
                                    uint8_t *value,
                                    esp_gatt_write_type_t write_type,
                                    esp_gatt_auth_req_t auth_req);
`

It seems I will need **gattc_if, conn_id (may be 0), handle, auth_req** to be provided. But I can't find a way to get it with a connected device in the source code provided. Worst case I will need to handle the whole BLE client process which should be double effort and may not worth it (only to solve this issue). May be a support API from espressif?

Thank you for reading and any support is appriciated.

xiewenxiang, 2019-11-14:
If you use AT, you don't need to worry about the API.  You can refer to the description of AT+BLEGATTCWR.

boyhuesd, 2019-11-15:
> 
> 
> If you use AT, you don't need to worry about the API. You can refer to the description of AT+BLEGATTCWR.

Hi @xiewenxiang ,
Thanks for the reply.
The problem I'm facing is (I believe) AT+BLEGATTCWR didn't write to the correct handle on my device.
The reason I think about that is firstly, I cannot succesfully writtern to my device UUID and I can do it by adding 1 to the characteristic handle when using other BLE MCU (I confirmed it on CC2541 and ST BlueNRG). 
I want to elaborate more on this based on my understanding. For example:

A normal process:
```
AT+BLEGATTCWR=0,3,4,,6
>
<data>
```
in which esp32-at will call:
```
esp_ble_gattc_write_char(gattc_if,
0, // con_id
**19, // handle, for example it is 19 which the stack discovered**
6, // value len
data,
write_type,
auth_req).
```
ESP32 AT will response back okay via uart, but actually my BLE device will not receive it.
Which leads me to believe the correct call would be:
```
esp_ble_gattc_write_char(gattc_if,
0, // con_id
** 20, //19 + 1 // handle, it is 19 which the stack discovered plus 1 for this particular device**
6, // value len
data,
write_type,
auth_req).
```

That is why I need supports on updating handle with current esp32-at firmware.
Thank you for reading and looking forward to your support.

closed 2020-01-13 by boyhuesd