espressif/esp-at #272: Compile for 1M size flash fail

Dangku, 2019-11-15:
CONFIG_IDF_TARGET_ESP8266=y
CONFIG_IDF_TARGET="esp8266"
CONFIG_ESPTOOLPY_FLASHSIZE_1MB=y
CONFIG_ESPTOOLPY_FLASHSIZE="1MB"
CONFIG_SPI_FLASH_SIZE=0x100000
CONFIG_PARTITION_TABLE_TWO_OTA=y
# CONFIG_PARTITION_TABLE_CUSTOM is not set

Generating esp8266.project.ld
LD build/esp-at.elf
/media/dangku/mywork/k210/source/esp8266/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/5.2.0/../../../../xtensa-lx106-elf/bin/ld: /media/dangku/mywork/k210/source/esp8266/esp/esp-at/build/esp-at.elf section `.flash.text' will not fit in region `iram0_2_seg'
/media/dangku/mywork/k210/source/esp8266/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/5.2.0/../../../../xtensa-lx106-elf/bin/ld: address 0x402a9ba4 of /media/dangku/mywork/k210/source/esp8266/esp/esp-at/build/esp-at.elf section `.flash.rodata' is not within region `iram0_2_seg'
/media/dangku/mywork/k210/source/esp8266/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/5.2.0/../../../../xtensa-lx106-elf/bin/ld: address 0x402a9ba4 of /media/dangku/mywork/k210/source/esp8266/esp/esp-at/build/esp-at.elf section `.flash.rodata' is not within region `iram0_2_seg'
collect2: error: ld returned 1 exit status

Thanks

xcguang, 2019-11-15:
The firmware is too large for 1MB flash.

Dangku, 2019-11-15:
Yes, esp-at.bin oversize app partition, So using the default sdkconfig must set  flash size >= 2MB,  Is there a minimal sdkconfig for 1MB flash? or a docs for firmware size optimization?

Thanks

xcguang, 2019-11-15:
If enable OTA, it is too difficult using 1MB flash. You can disable some AT command and `make SILENCE=1` to try it.

closed 2019-11-15 by Dangku

hallard, 2020-02-22:
@Dangku did you succeeded  to have latest version running on 1MB flash (even without OTA)?

jandrassy, 2020-02-22:
@hallard use the latest master. there is a new option

hallard, 2020-02-23:
Awesome, I'll try that, to be honest I succedeed to compile the 2MB version but when flashed on ESP Wemos Mini D1 the code never started and stopped after boot message, I'll try this one thank

hallard, 2020-02-24:
@jandrassy just tried without success, more logs at #351