espressif/esp-at #273: Less range in AT+RFPOWER in new release v2.0.0.0_esp32

carlessole, 2019-11-15:
Hello,
I see that the way RFPOWER<wifi_power> is set has changed from an ENUM way ([v1.2.0.0](https://github.com/espressif/esp-at/releases/tag/v1.2.0.0)) to an ANALOG way ([v2.0.0.0_esp32](https://github.com/espressif/esp-at/releases/tag/v2.0.0.0_esp32)):
**[v1.2.0.0](https://github.com/espressif/esp-at/releases/tag/v1.2.0.0) (range is -1dBm to 19.5 dBm):**
[image]
**[v2.0.0.0_esp32](https://github.com/espressif/esp-at/releases/tag/v2.0.0.0_esp32)(range is 10dBm to 20.5 dBm):**
[image]

Is there a way to set <wifi_power>to a lower value in ([v2.0.0.0_esp32](https://github.com/espressif/esp-at/releases/tag/v2.0.0.0_esp32) than 10 dBm.

Thanks

xcguang, 2019-11-16:
Please refer to https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Factory_Parameter_Bin.md

carlessole, 2019-11-18:
I have readen this document but I still don't know how it can solve my problem.
In tha Factory Param Type table there is just the _tx_max_power_ column, nothing else related with the wifi power.
So, I don't understand how to set a lower value than 10 dBm in [v.2.0.0.0_esp32](https://github.com/espressif/esp-at/releases/tag/v2.0.0.0_esp32)
Thanks

xcguang, 2019-11-19:
For example, if you want to set 9dBm, you can configure `tx_max_power` 36. But you can't set it by AT command. BTW, Do you need such command?