espressif/esp-at #275: Change debug and AT console to UART0 failed

Dangku, 2019-11-18:
Hi,

I follow https://github.com/espressif/esp-at/blob/master/docs/How_To_Set_AT_Port_Pin.md to change  my debug and AT port to UART0, only debug port is changed, but AT port is still io-15/io-13, here is my step:

1. make menuconfig to set defualt debug port to uart0, set AT uart configuration
`CONSOLE_UART_DEFAULT=y`
`CONFIG_AT_UART_DEFAULT_PARITY_BITS=0`
`CONFIG_AT_UART_DEFAULT_FLOW_CONTROL=0`

2. update facotry_param_data.csv to set TX/RX io
`PLATFORM_ESP8266,WROOM-02,0xfcfc,1,0,78,1,13,CN,115200,1,3,-1,-1,-1,-1`

3. Makefile platform and module
`export ESP_AT_PROJECT_PLATFORM ?= PLATFORM_ESP8266`
`export ESP_AT_MODULE_NAME ?= WROOM-32`

4. compile
`$ make`
`$ make flash`

I add some debug message to main/interface/uart/at_uart_task.c, and it shows that factory_para partition is read successfully, but all data is 0xFF.

`at_uart_init
factory_param partition read successful
data[0]=ff, data[1]=ff, data[16]=ff, data[17]=ff, data[18]=ff, data[19]=ff
tx_pin=15, rx_pin=13
2.0.0
ready`

Thanks

xcguang, 2019-11-18:
```
export ESP_AT_PROJECT_PLATFORM ?= PLATFORM_ESP8266
export ESP_AT_MODULE_NAME ?= WROOM-32
```

if it is `ESP8266 platform`, please change `ESP_AT_MODULE_NAME ` to `WROOM-02`

Dangku, 2019-11-18:
[吐血].gif

thanks

closed 2019-11-18 by Dangku