espressif/esp-at #276: [ESP8266] Can't compile project - undefined references

miguel9554, 2019-11-19:
I'm following the compiling instructions described [here](https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Get_Started.md#esp8266-platform), but when I run `make flash` I get the following:

```
LD build/esp-at.elf
/home/miguel/esp/esp-at/components/at/lib/libesp8266_at_core.a(at_port.o):(.literal.at_recv_task+0x38): undefined reference to `xQueueSelectFromSet'
/home/miguel/esp/esp-at/components/at/lib/libesp8266_at_core.a(at_port.o):(.literal.esp_at_module_init+0x20): undefined reference to `xQueueCreateSet'
/home/miguel/esp/esp-at/components/at/lib/libesp8266_at_core.a(at_port.o):(.literal.esp_at_module_init+0x24): undefined reference to `xQueueAddToSet'
/home/miguel/esp/esp-at/components/at/lib/libesp8266_at_core.a(at_port.o):(.text.at_recv_task+0x4c): undefined reference to `xQueueSelectFromSet'
/home/miguel/esp/esp-at/components/at/lib/libesp8266_at_core.a(at_port.o): In function `esp_at_module_init':
(.text.esp_at_module_init+0x74): undefined reference to `xQueueCreateSet'
/home/miguel/esp/esp-at/components/at/lib/libesp8266_at_core.a(at_port.o): In function `esp_at_module_init':
(.text.esp_at_module_init+0x88): undefined reference to `xQueueAddToSet'
/home/miguel/esp/esp-at/components/at/lib/libesp8266_at_core.a(at_port.o): In function `esp_at_module_init':
(.text.esp_at_module_init+0x98): undefined reference to `xQueueAddToSet'
collect2: error: ld returned 1 exit status
make: *** [/home/miguel/esp/ESP8266_RTOS_SDK/make/project.mk:510: /home/miguel/esp/esp-at/build/esp-at.elf] Error 1
```

I've succesfully installed the toolchain as instructed [here](https://docs.espressif.com/projects/esp8266-rtos-sdk/en/v3.2/get-started/linux-setup.html), and downloaded and seted up `ESP8266_RTOS_SDK` as instructed [here](https://docs.espressif.com/projects/esp8266-rtos-sdk/en/v3.2/get-started/index.html#get-esp8266-rtos-sdk).

I've managed to succesfully compile and flash the `project_template` example, but then I failed trying to compile the `ESP-AT` project. However, after failing the flashing of the `ESP-AT`, if I try to flash the `project_template` I get the following error:

```
/home/miguel/esp/ESP8266_RTOS_SDK/components/mbedtls/mbedtls/library/ssl_ticket.c: In function 'mbedtls_ssl_ticket_write':
/home/miguel/esp/ESP8266_RTOS_SDK/components/mbedtls/mbedtls/library/ssl_ticket.c:337:17: error: implicit declaration of function 'mbedtls_cipher_auth_encrypt' [-Werror=implicit-function-declaration]
     if( ( ret = mbedtls_cipher_auth_encrypt( &key->ctx,
                 ^
/home/miguel/esp/ESP8266_RTOS_SDK/components/mbedtls/mbedtls/library/ssl_ticket.c: In function 'mbedtls_ssl_ticket_parse':
/home/miguel/esp/ESP8266_RTOS_SDK/components/mbedtls/mbedtls/library/ssl_ticket.c:428:17: error: implicit declaration of function 'mbedtls_cipher_auth_decrypt' [-Werror=implicit-function-declaration]
     if( ( ret = mbedtls_cipher_auth_decrypt( &key->ctx, iv, 12,
                 ^
/home/miguel/esp/ESP8266_RTOS_SDK/components/mbedtls/mbedtls/library/ssl_ticket.c: At top level:
cc1: warning: unrecognized command line option '-Wno-frame-address'
cc1: some warnings being treated as errors
make[1]: *** [/home/miguel/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:292: mbedtls/library/ssl_ticket.o] Error 1
make: *** [/home/miguel/esp/ESP8266_RTOS_SDK/make/project.mk:571: component-mbedtls-build] Error 2
```
Removing `ESP8266_RTOS_SDK` and cloning the repo again solves this issue, but I still can't flash the `ESP-AT` project.

xcguang, 2019-11-19:
So strange, is ESP8266_RTOS_SDK cloend auto when compiling?
You can try `make clean` and ` rm -rf build sdkconfig`, then `make defconfig` and `make` to recompile, notes: please download SDK manually.

miguel9554, 2019-11-19:
Yes, indeed the repo was being cloned. I modified `module_config/module_esp8266_default/IDF_VERSION` to point to the release version 3.2 (commit `3ad3fd783da936984b4247e0974a1a8496796346`), and now the repo is not being cloned, but I'm getting the following error when running `make flash`:

```
/home/miguel/esp/esp-at/main/interface/uart/at_uart_task.c: In function 'at_uart_init':
/home/miguel/esp/esp-at/main/interface/uart/at_uart_task.c:322:5: error: too many arguments to function 'uart_driver_install'
     uart_driver_install(esp_at_uart_port, 2048, 2048, 10,&esp_at_uart_queue, 0);
     ^
In file included from /home/miguel/esp/esp-at/main/interface/uart/at_uart_task.c:37:0:
/home/miguel/esp/ESP8266_RTOS_SDK/components/esp8266/include/driver/uart.h:430:11: note: declared here
 esp_err_t uart_driver_install(uart_port_t uart_num, int rx_buffer_size, int tx_buffer_size, int queue_size, QueueHandle_t *uart_queue);
           ^
make[1]: *** [/home/miguel/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:286: interface/uart/at_uart_task.o] Error 1
make: *** [/home/miguel/esp/ESP8266_RTOS_SDK/make/project.mk:517: component-main-build] Error 2
```

I tried what you suggested, and the same errors appear.

xcguang, 2019-11-19:
Please donot modify any code and try it. You can't change SDK version at will

jandrassy, 2019-11-19:
did you change the Makefile for esp8266?

miguel9554, 2019-11-20:
Ok, I will try it. What do you mean with downloading SDK manually? not cloning the repo?

xcguang, 2019-11-20:
Yes, it will auto be cloned when `make`, you only need to change Makefile as following,
```
export ESP_AT_PROJECT_PLATFORM ?= PLATFORM_ESP8266
export ESP_AT_MODULE_NAME ?= WROOM-02
```

miguel9554, 2019-11-20:
I did it, and I'm getting the same error

xcguang, 2019-11-20:
Please give your Makefile in esp-at project.

miguel9554, 2019-11-25:
Sorry for the delay, here they are. I added the txt extension for github to allow me to upload, they don't have the extension on my system.

[Makefile.txt](https://github.com/espressif/esp-at/files/3887803/Makefile.txt)
[sdkconfig.txt](https://github.com/espressif/esp-at/files/3887805/sdkconfig.txt)

xcguang, 2020-01-09:
Please try it using a5a8c741ffa0ad504321d078a2e7d70b6cd47afb

xcguang, 2020-01-09:
If the issue persists, please reopen it.

closed 2020-01-09 by xcguang