espressif/esp-at #281: esp32 AT command doesnot send

reshma26p, 2019-11-22:
hello..
firstly i worked on esp32 at it works very well after that i erase flash and then make flash same..then AT commands does not transfer

xiewenxiang, 2019-11-23:
Hello @reshma26p 

Can you give a more detailed description..

reshma26p, 2019-11-25:
hello.. @xiewenxiang 
i am using esp wroom32 board..i want to bluetooth control using AT commands..i want to use same tx rx pin for flash and to AT commands so as  given in ESP_AT_Get_Started.md and How_To_Set_AT_Port_Pin.md i make settings but AT commands does not get tx.
before some days i was working on same it works very well after that i make erase flash and then same data flash and problem is now

closed 2019-11-26 by reshma26p