espressif/esp-at #283: ERR CODE:0x010b0000 and "busy p..." when configuring SSL with AT+CIPSSLCCONF

tvandergeer, 2019-11-25:
During the init phase of my ESP32 I run a dozen commands. Right after setting the configuration of the NTP server with `AT+CIPSNTPCFG` and getting the current time with `AT+CIPSNTPTIME?` I'm setting the SSL config for link ID 4 to auth mode 3. The `AT+CIPSSLCCONF` command returns with an error and a "busy p..." followed by an OK. 

```
AT+CIPSNTPCFG=1,0,"2.nl.pool.ntp.org"

OK
AT+CIPSNTPTIME?
+CIPSNTPTIME:Mon Nov 25 20:39:19 2019
OK
AT+CIPSSLCCONF=4,3,0,0
ERR CODE:0x010b0000

busy p...

OK
```

I'm not sure whether the SSL connections are configured correctly.

What's going on?

xcguang, 2019-11-28:
Hi,
It looks the command you send is correct. 
Maybe there are other blank characters following `AT+CIPSSLCCONF=4,3,0,0\r\n`, and it will cause the error code and a "busy p..."

xcguang, 2020-01-14:
Hi,
Is the issue solved?

tvandergeer, 2020-01-14:
I haven't tested yet with the most recent version of the firmware. Did you fix anything in this respect?

ustccw, 2020-01-14:
@tvandergeer i am not sure it there some wrong input AT command in your side.
it looks work well here.
[image]

tvandergeer, 2020-01-14:
Ok, that looks slightly better. There's still an error though. Any idea why that is?

ustccw, 2020-01-14:
@tvandergeer cause here is no a valid link, which linkID=4

tvandergeer, 2020-01-14:
@ustccw But this config should be set prior to setting up a link. As per [documentation](https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md#cmd-SSLCCONF):
> Call this command before establish SSL connection if you want configuration take effect immediately.

ustccw, 2020-01-14:
@tvandergeer i am so sorry for my wrong reply.

user should enable the multi-link firstly by `AT+CIPMUX=1`
then to configure, like the following: 

[image]

tvandergeer, 2020-01-14:
@ustccw Ah, clear! Looks good. Will give it a try and report back

xcguang, 2020-02-07:
Hi,
Is the issue solved? Can I close it?

closed 2020-03-12 by xcguang