espressif/esp-at #284: Problem with connecting to AWS MQTT broker using certificate

Maher9494, 2019-11-28:
Hello, 
I'm testing the new version of ESP-AT 2.0.0, My aim is to connect to amazon AWS server via MQTT using private key and a certificate. 
I have followed the AWS instructions and created the thing on AWS. 
the commands that I have issued are the following 
`AT+MQTTUSERCFG=0,4,"ESP32","","",0,0,"" `
response is OK 
`AT+MQTTCONN=0,"xxxxxxxxx-ats.iot.us-west-2.aws.amazon.com",8883,0`
the response is: +MQTTDISCONNECTED:0
ERROR 
knowing that I have already put my certificates instead of the default generated mqtt_cert.bin and mqtt_key.bin and the mqtt_ca.bin
and then I have `make factory_bin` which was uploaded to the ESP32.

xcguang, 2019-11-29:
We have a guidance about this, but it is in Chinese, https://blog.csdn.net/espressif/article/details/101714951.

Maher9494, 2019-11-29:
Thanks A lot I have followed it and it has worked :)

closed 2019-11-29 by xcguang