espressif/esp-at #285: AT port

reshma26p, 2019-11-29:
hello,
i am using esp32 wroom board
by using separate port as to flash gpio1,3 and for ATgpio16,17 working well 
 i want to use same port for flash code which is of UART0 AND pin tx gpio1 ,rx gpio3 and AT port which is given that at UART1 of pin tx gpio17 ,rx gpio16 in https://github.com/espressif/esp-at  but actually in wroom32  datasheet pin tx gpio17,rx gpio16 of UART2.
As given in https://github.com/espressif/esp-at/blob/master/docs/How_To_Set_AT_Port_Pin.md
i made that setting 
make menuconfig > Component config > ESP32-specific > UART peripheral to use for console output (0-1) > UART1
their is UART0 UART1 option but actually from data sheet gpio16,17 which is of UART2
SO THAT AT COMMANDS DOES NOT GET TX

xcguang, 2019-11-29:
The pins for UART can be configured as others. Please refer to the chapter about IO_MUX  and GPIO in the datasheet and `esp32 technical reference manual`.

reshma26p, 2019-11-29:
YES ..but in this case how to change...i trying for change as procedure in https://github.com/espressif/esp-at/blob/master/docs/How_To_Set_AT_Port_Pin.md
make menuconfig > Component config > ESP32-specific > UART peripheral to use for console output (0-1) > UART1
NO OPTION OF UART2

reshma26p, 2019-11-29:
@xcguang please give me response its urgent for me

xcguang, 2019-11-29:
Why UART2?   
The port sending AT command is UART1.  gpio16,17 are not only for UART2, you can configure it for UART0 or UART1.

reshma26p, 2019-11-29:
yes i am done settings as given but commands does not get tx instead  of i am using separate pin without change in menuconfig as gpio1,3 to flash and gpio16,17 for AT commands is working well

xcguang, 2019-11-29:
[image]
Do you modify them? If so, Please `make clean & make`

reshma26p, 2019-11-29:
yes..i did all  procedure given in https://github.com/espressif/esp-at/blob/master/docs/How_To_Set_AT_Port_Pin.md
  
  
  

----------------------------------------

SONGjiaxiu, 2019-11-29:
Set GPIO1(TX) and GPIO3(RX) to be both the log pin and AT port pin according to the document`How_To_Set_AT_Port_Pin.md`, it works well.

[image]

closed 2019-11-29 by reshma26p