espressif/esp-at #286: AT BT

reshma26p, 2019-11-29:
is it possible to AT commands of BT giving from BT port?

reshma26p, 2019-11-29:
please give me response

xcguang, 2019-11-29:
turn on bluetooth by power on and available to discover and connectable
---- The default firmware does not support it

change bluetooth name and settings of baud rate ,parity by AT commands from bluetooth COM port
----- Please refer to the AT command document

reshma26p, 2019-11-30:
i want to change name , baud rate from COM port which is getting by bluetooth pairing i.e bluetooth port

xcguang, 2019-11-30:
Now have no such command, maybe you can customize it.

reshma26p, 2019-11-30:
how?

xiewenxiang, 2020-03-19:
SPP transmission has no baud rate concept and is completely dependent on the actual transmission speed of the bluetooth chip, which can be understood as 3Mbps.

If you want to set device name , you can use comand `AT+BTNAME`

closed 2020-03-19 by xiewenxiang