espressif/esp-at #287: MQTT

dani-fc, 2019-12-02:
Hi,

i use AT command to connect to mosquitto test broker (test.mosquitto.org).
When on a topic is published a message of 11 or 13 or 15 byte, after that message is received, ESP32 always reset. ESP32 sometimes receive correctly message of 12 or 14 byte.

I have try with MQTT over TCP and MQTT over WebSocket Secure(based on TLS, no certificate verify).

ESP-AT version 2.0.0 with MQTT AT Commands enabled.
ESP_IDF version 3.3

ustccw, 2019-12-24:
@dani-fc thanks for your feedback.
we have fixed it internally. It will be synced to GitHub soon.

Sorry for the Inconvenience.

ustccw, 2020-01-09:
@dani-fc please try with the latest version, it had fixed.

dani-fc, 2020-01-09:
@ustccw thank you, now with latest version it works fine

closed 2020-01-10 by xcguang