espressif/esp-at #290: AT+CWJAP? and AT+CWLAP response is not OK

carlessole, 2019-12-03:
Hi,
When setting the `AT+CWJAP`, escape the strings is mandatory:

> Escape character syntax is needed if SSID or password contains any special characters, such as , or " or \.

But when asking `AT+CWJAP?`, the ESP32 response is not escaped, and so, there is no proper way to parse all the fields properly (if MySSID contains "," for example). It must be an error right?

And regarding the documentation the resposne just have 4 fields but I see that it has 7 fields

`+CWJAP:"MySSID","e0:28:6d:8a:d0:1a",11,-48,0,0,0
`

Which fields are them?

The same happens with `AT+CWLAP`:
```
AT+CWLAP
+CWLAP:(3,"1\1"1,2\\2""2,,",-40,"e0:28:6d:8b:54:9c",9)
```
The SSID is not escaped.

Thanks

ustccw, 2019-12-24:
@carlessole The fields in the response of "AT+CWJAP" could be the explanation: 
```
<cmdname>:<"ssid">,<"bssid">,<channel>,<rssi>,<station_configured_authmode_gt_WIFI_AUTH_WPA_PSK_flag>,<beacon_timeout_reconnect_flag>,<station_configured_listen_interval>
```

Sorry for not updating the document in time

ustccw, 2019-12-24:
@carlessole the escape character is for AT command to identify.
if user want to parse the response, could try to parse response from string tail to string head.

closed 2021-04-14 by FayeY