espressif/esp-at #292: ESP-AT-SPI firmware on ESP32-D2WD

YutingYou, 2019-12-04:
Hi ESP32,
I'm trying the AT SPI firmware on ESP32-D2WD, but I got blocked by the hardware configuration.
> It should be noted that CMD and DATA lines D0-D3 of the slave should be pulled up by 50KOhm resistor even when there is no connection. In addition, the voltage selection efuse should be written as 3.3v. For specific notes, please refer to SD Pullup Requirements

Since the voltage of the internal flash of ESP32-D2WD is 1.8V, can I change the efuse to 3.3V, if I can't, what should I do?

xcguang, 2019-12-05:
Please refer to https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/sd_pullup_requirements.html

YutingYou, 2019-12-05:
@xcguang  Thanks for the reply, I've got it worked on my ESP32-D0WD, now I'm trying to run AT SPI firmware on ESP32-D2WD which has internal 2MB flash, however, the flash gets overlapped at address: 0x20000:
```
Flashing binaries to serial port /dev/cu.SLAB_USBtoUART (app at offset 0x30000)...
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x20000 for file: /Users/yuting/esp/esp-at/build/at_customize.bin
make: *** [flash] Error 2
```

As per the `module_config/module_esp32_default/at_customize.csv` configuration, the flash from 0x20000 to 0x100000 is occupied by `at_customize.bin`,  while the `esp-at.bin` needs 1.4MB, here're my questions:
1. Can the AT SPI firmware work on ESP32-D2WD?
2. Can I remove some components to decrease flash usage?

xcguang, 2019-12-06:
1. The latest AT SPI is just support ESP8266; but you can use SDIO AT on ESP32 Slave, and SPISD on host via SPI hardware.
2. You can disable BT and BLE command by `make menuconfig`

closed 2019-12-09 by YutingYou