espressif/esp-at #293: AT+BLECONN

thuLssss, 2019-12-05:
Esp32 client can establish connection with "AT+BLECONN=<conn_index>,<remote_address>" and decide to connect which device according to remote_address. I want to know if Esp32 client can decide to connect which device according to the device name?

alexdecei, 2019-12-18:
I think you can find the device name in the advertisment packet. To get this data, you must scan first, then decode the lines, for example:

```
AT+BLESCAN=1

OK
+BLESCAN:23:d7:d3:80:c5:1c,-82,1eff060001092002ea1cbbc34f2353c946e185f83dd60665dcaa4d3396771b,,1
+BLESCAN:3c:5f:ca:fc:63:db,-90,1eff06000109200268b7a330ea7707182a069f0f714ca5b7112320c830bd32,,1
+BLESCAN:dc:0d:30:00:00:98,-83,0201061aff4c000215fda50693a4e24fb1afcfc6eb07647825275165c1fd,,0
+BLESCAN:35:a1:e2:51:58:94,-89,1eff060001092002aab677e25754f3189fd0fc11f8009b5d044b97ed391791,,1
```

To decode/analyse, I don't know if you can do that with AT commands.

xiewenxiang, 2020-01-09:
The remote device must be needed in broadcast

Also note the address type, if the remote device is a random address type, you need to add optional parameters

```c
AT+BLECONN=0,"5b:f5:b9:de:b2:65",1
```

xcguang, 2020-02-07:
Hi,
Is the issue solved? Can I close it?

closed 2020-03-15 by xcguang