espressif/esp-at #294: Not receiving `+IPD` messages after a `LINK_CONN` as server when in passive mode

tvandergeer, 2019-12-06:
In my case test case I'm using `AT+CIPRECVMODE=1` (aka "passive mode"). When I setup a softAP and listen (as a server) for connections on port 80 I do receive the `+LINK_CONN` messages for an incoming connection (from an http client, e.g curl or a browser), but it does not send any `+IPD` messages for incoming data.

It does work with `AT+CIPRECVMODE=0`, but that's not what I want to use. It also works when I connect with telnet to port 80, wait a couple of seconds and then send data manually. It seems that the ESP in passive mode doesn't signal data, that's coming in directly after connecting, with a `+IDP` message.

Another observation is that when I do a `AT+CIPRECVLEN?` directly after receiving a `LINK_CONN` the ESP does indicate that there's data available:
```
+LINK_CONN:0,0,"TCP",1,"192.168.4.2",49469,80
AT+CIPRECVLEN?
+CIPRECVLEN:75,-1,-1,-1,-1

OK
AT+CIPRECVDAT,=1,75
to read 75 bytes
+CIPRECVDATA:75,"192.168.4.2",49478,GET / HTTP/1.1
Host: 192.168.4.1
User-Agent: curl/7.54.0
Accept: */*

OK
```

The expected behaviour is that (in passive mode) the following message follows the `LINK_CONN` line:
```
+IPD,1,75
```

I'm using an ESP32 with the latest firmware compiled from the Master branch.
```
AT+GMR
AT version:2.1.0.0-dev(0ad1446 - Dec  3 2019 08:19:15)
SDK version:v3.3
compile time(71f8029):Dec  6 2019 13:21:51
Bin version:2.0.0(WROOM-32)

OK
```

xcguang, 2020-03-15:
Hi,
Can you try the latest master branch? We have optimized some process.

tvandergeer, 2020-05-04:
I've successfully tested this with latest code on Master:
```
AT version:2.1.0.0-dev(eaa3743 - Apr 30 2020 03:40:57)
SDK version:v4.0-327-gfe67bedee
compile time(aa30192):May  2 2020 14:56:15
Bin version:2.0.0(WROOM-32)

OK
```
It's now showing ```+IPD``` messages right after a +LINK_CONN message. Below is an example of a request from a browser on my mobile phone where I reloaded a webpage a few times (without sending a response from the ESP32)
```
+LINK_CONN:0,0,"TCP",1,"192.168.4.2",57494,80

+IPD,0,373
+LINK_CONN:0,1,"TCP",1,"192.168.4.2",57495,80

+IPD,1,373
+LINK_CONN:0,2,"TCP",1,"192.168.4.2",57496,80

+IPD,2,373
+LINK_CONN:0,3,"TCP",1,"192.168.4.2",57497,80

+IPD,3,373
```

closed 2020-08-18 by xcguang