espressif/esp-at #295: [ESP8266] command ending of \r\n

VaZso, 2019-12-09:
Hi,

I see there is an option in "menuconfig" called AT command terminator support where I can specify a termination character (however, only one character).

If I disable this option, it seems to accept \r\n line ending but then it continues to accept with \r till a point where it also need to input \r\n then the same again.

I would like to use \r\n in all the commands but it seems it does not work.
How could I do this or is there any progress to get this function working?

It seems the source code of this part is not available so I can not modify its behaviour.

Thank you in advance.

VaZso, 2019-12-11:
I see there is a function called to set terminal character which is limited to only one character:
This one:
> esp_at_custom_cmd_line_terminator_set(uint8_t* terminator);

However, it says the following in comment:
> Set AT command terminator, by default, the terminator is "\r\n"

I don't want a special character for termination but would like to use "\r\n", however, that is what not working properly, the library only works well by calling the function above to use an own terminal character like "\n" and use commands using the specified terminal character.

So, when it is set to use the default "\r\n", then it accepts one command well with "\r\n" at the end (like calling "AT\r\n") but not the second or third one. If I write "AT\n" to the second / third one then it accepts it but it is not an expected behaviour. It should only work if I write "AT\r\n" again but it causes an ERROR response.

Also, if I write another command instead of "AT", I have to use "\r\n" again.
So the handling of "\r\n" characters are not consequent.

Do others have it working well or it is really broken in the closed library?

xcguang, 2019-12-11:
Hi,
Currently, `esp_at_custom_cmd_line_terminator_set` just supports one character.

VaZso, 2019-12-11:
Hi,

I see, but default behaviour seems to be using \r\n which does not really work.

Am I miss something?

VaZso, 2019-12-16:
Do I misunderstood that \r\n is the default behaviour?
Should I except any change / repair in the near future?

xcguang, 2020-01-09:
[QCOM.zip](https://github.com/espressif/esp-at/files/4038778/QCOM.zip)
Could you try this tools? Please choose `Send With Enter` option.

xcguang, 2020-02-07:
Hi,
Is the issue solved? Can I close it?

VaZso, 2020-02-07:
Sorry, I had a lot of work in last weeks so I had a really few spare time.

As far as I can see, QCOM is a terminal program (?) - I haven't run it yet as I use Linux so I may try it in virtual machine or another machine, but I currently not see how it helps.
What do you like me to test?

Am I understand well that accepting \r\n should be the normal behaviour?
When selecting "Send With Enter" option, what will it send? \r\n or \n ?

I should prepare my hardware / software environment to test it but it would be good to know what to test and what is the expected behaviour.

Thank you in advance.

xcguang, 2020-02-07:
What do you like me to test?
------- As you said, default behaviour seems to be using \r\n which does not really work, I think maybe there is something wrong with the tools, because we and other users can works well. So I suggest you can confirm it.

Am I understand well that accepting \r\n should be the normal behaviour?
---- Yes, you are right, it should end with \r\n

When selecting "Send With Enter" option, what will it send? \r\n or \n ?
----- It is  \r\n

xcguang, 2020-02-07:
Or you can send "AT\r\n" on linux, and capture the data on the UART cable to confirm it.

VaZso, 2020-02-07:
Do I need to update source code and compile it again (it is almost two months old if I am right)?
So is there anything potentially changed in the blob related to my problem?

(Currently my hardware is not where I am anyway.)

Also, I can confirm the original firmware of ESP8266 worked as expected (\r\n) even after reflashed it but with the compiled code, it did what I wrote above.

I have tried it under Linux the same way as it worked with original firmware and I have stopped working on it when I saw this version has this problem... also another project took priority then and the project which I wanted to replace stock firmware in is currently using the stock one and after month(s) it is not really having much chance to replace it this state...

I will try it soon anyway but I know this behaviour was caused by something in the firmware and not the operating system or a way I communicate with the module.

xcguang, 2020-02-08:
[image]

Did you enable `AT command terminator support`? If not, it should be same as the original firmware.

Can you provide the command response you got?

VaZso, 2020-02-10:
No, I left that option disabled as it only allows one character as terminator.

Now I have compiled a whole new project based on latest sources.
xtensa comes from [here](https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/linux-setup.html) (actually [this one](https://dl.espressif.com/dl/xtensa-lx106-elf-linux64-1.22.0-100-ge567ec7-5.2.0.tar.gz)) and esp-at is the latest.

I had to modify WROOM-02 line of factory_param_data.csv to this one:
> PLATFORM_ESP8266,WROOM-02,0xfcfc,1,0,78,1,13,CN,115200,1,3,13,15,-1,-1

Everything else is as default, including "AT command terminal support" as disabled.

After uploaded this firmware, I have connected to the module using az USB-UART converter by using the following command:
> screen /dev/ttyUSB3 115200

Here I can enter "AT\r\n" where it answers "OK"
Then I can enter "AT\r" then it instantly answers "OK".
I can enter "AT\r" several times, it always answers "OK" before I could enter \n
If I enter "AT\r\n" then "AT\r\n" I receive an "ERROR" because of the \n in front of "AT"
...but when I enter "AT\r\n" again, I receive "OK", then it only accepts "AT\r" again as above.

Based on that, I have generated a file which has AT

>

VaZso, 2020-02-10:
I was thinking over a bit and I think it is a bug, so it should not accept "AT\r\nAT\rAT\rAT\r" -like strings because if the communication is slow between ESP and another microcontroller, it can cause problems.

So it should always wait for "\r\n" as terminal string and not handle it just when entering "\r" while it waits for "\r\n" in other cases.

xcguang, 2020-02-11:
Yes, it will wait "\r\n", then to process it.
If you send "AT\r", it should have no answer, and will wait "\n", until the length of the strings reaches 256. 

>Then I can enter "AT\r" then it instantly answers "OK".
I can enter "AT\r" several times, it always answers "OK" before I could enter \n

Does the `screen` tool add '\n' automatically following '\r'? If not, please provide your firmware.

VaZso, 2020-02-24:
Sorry for late reply, I had to work extensively on a project...

Yes, it waits for "\r\n" then it starts to accept it even if it receives "\r" but not receives "\n" immediately... but that case "\n" goes to the front of next message, so next message will fail.

With the program "screen", I can send "\r" or "\n" characters separately.
It does not send additional "\n" characters and also it works well using original ESP firmware where "\r\n" works as expected.

The difference is this sends character just when it pressed on keyboard so "\r" and "\n" does not come together this case.

That is why I have tried to send these messages from files so to ensure they go through at once, without further delay between characters.

Side note: Linux terminal (tty) has a traditional behaviour of converting "\r" to "\n" but it does not affect ttyUSB devices and can be switched off on ttyACM devices and also screen turns it off.
This case it is a ttyUSB device.

Here is my [esp-at.bin](https://github.com/espressif/esp-at/files/4243663/esp.zip)

VaZso, 2020-03-12:
I don't know if there were any modifications related the default "\r\n" handling we discussed above but the current code works for me as expected, so I think it was resolved in code base.

As it is resolved now, I close this issue.

Thank you.

closed 2020-03-12 by VaZso