espressif/esp-at #296: [ESP8266] Lack of AT+CWJAP_CUR

VaZso, 2019-12-09:
Hi,

It seems esp-at currently does not support AT+CWJAP_CUR command.
AT+CWJAP works well, but that way it saves the settings and next time when the module reboots, it automatically connects to that network.

I don't want that save to happen, instead, I would like it to only connect the network when I give appropriate details together with AT+CWJAP_CUR.

Also, I don't want to write the flash over and over again when I specify another network.

Is there any solutions for that?
Why esp-at seems to not support the simplest "connect but not save" command?

I don't think it is a complicated thing to include this feature but as it seems the source code of main functions are not available, I can not do that.

Thank you your help in advance.

jandrassy, 2019-12-09:
there is `AT+CWAUTOCONN` to set if the esp should connect automatically.

but I agree, that _CUR commands would be very useful

VaZso, 2019-12-09:
Thank you.

However, if AT+CWJAP saves the data regardless of AT+CWAUTOCONN, then I will have to stick on another firmware as I don't want to worn out the flash in long-term use.

I really don't like the idea of writing its own flash if is is not necessary - and as implementing AT+CWJAP_CUR does not look to be a complicated thing, I don't know why it is still not implemented.

VaZso, 2019-12-16:
Nothing of the developers who have access to the source code?
...is that hard to remove flash write from AT+CWJAP and name it as AT+CWJAP_CUR?

Are there anyone relevant reading this?
Should I forget this whole esp-at project?

xcguang, 2019-12-17:
Currently, it does not support `AT+CWJAP_CUR?`

VaZso, 2019-12-17:
Correct, it currently does not support it - I have compiled esp-at and tried to work with it but stopped at this point.

It is not even written in ESP_AT_Commands_Set.md

xcguang, 2019-12-17:
[image]

You can try disable this option

VaZso, 2019-12-17:
Thank you - it may be a workaround, but a custom command of AT+CWJAP_CUR should be implemented then which forwards the request somehow to AT+CWJAP

Currently it seems at least it does not save data to flash.

xcguang, 2020-03-12:
Hi,
We have added AT+SYSSTORE, https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md#Basic-AT, maybe it can match your requirement.

VaZso, 2020-03-12:
Hi,

Thank you.

I have compiled and upload it to my project and I see this command and it seems to work, so AT+CWJAP is not saved.

However, I should implement AT+CWJAP_CUR and similar commands in order to be compatible with the old firmware and check the state of this command somehow...

However, now I see my other problem with "\r\n" handling is also resolved or at least it works now.

VaZso, 2020-03-12:
Hi,

Is there an official way to communicate with ESP-AT firmware inside  an own task?
...so I would like to modify +SYSSTORE value and rewrite AT+CWJAP_CUR to AT+CWJAP (and also some other commands from _CUR to the same CMD without cur).

So I would like to solve my problem of non-existing _CUR commands which is why this issue was opened...

Thank you in advance.

xcguang, 2020-03-12:
You can define your command, in the api, calling `esp_wifi_set_storage(WIFI_STORAGE_RAM);` but I think it is not a good way. 

Or you change `AT+CWJAP_CUR` into `AT+SYSSTORE=0` and `AT+CWJAP_CUR`. All input data is from `at_port_read_data`, and all output data is to `at_port_write_data`

VaZso, 2020-03-12:
I thought of calling AT+SYSSTORE=0 at init time because I basically don't want ESP-AT to save data to flash as I would like it to be handled by another MCU and I am also afraid of ESP-AT may write flash too often as its source code is unavailable.

As of _CUR-type CMDs, it seems ESP-AT project don't want to implement them to maintain compatibility if I understood well.

So I may need to create a handler in-between serial communication handlers to do this if I am right...

xcguang, 2020-03-22:
Yes, you are right

VaZso, 2020-03-22:
So if you have the source code of ESP-AT, this modification should be easy.
As we don't have the source code, modification is much more complicated compared the above.

Also, ESP-AT is incompatible with original firmware because of the lack of such instructions.

...but the project is not willing to do this modification...

Cool...

xcguang, 2020-03-22:
Sorry for that now we have no plan to add _CUR command, and our company policy does not permit us to open the source code.

VaZso, 2020-03-22:
Thank you your reply.

It is not a really good news anyway as I would need to also modify parts which belong to the project and on every update, I should do it again.

Basically it affects initialization and I should do another layer which calls esp_at_port_recv_data_notify function which may do a callback to get_data_length then a callback of read_data for own commands while I should handle original calls / callbacks and give it back to original handlers.
However, I don't understand what 'len' parameter I should pass to esp_at_port_recv_data_notify function if it does a callback to get length (there is not really a description of the real mechanism).

So it seems I will forget about this project for a while, then we will see it... :(

closed 2021-04-14 by FayeY