espressif/esp-at #298: esp-at 编译报 undefined reference to `xQueueAddToSet'

hzm476172557, 2019-12-11:
完全按照说明文档编译esp-at，最后链接阶段报undefined reference to `xQueueAddToSet'
怎么解决呢

hzm476172557, 2019-12-11:
LD build/esp-at.elf
C:/Users/Chris/Documents/work/ESP/msys32/home/Chris/esp/esp-at/components/at/lib\libesp8266_at_core.a(at_port.o):(.literal.at_recv_task+0x38): undefined reference to `xQueueSelectFromSet'
C:/Users/Chris/Documents/work/ESP/msys32/home/Chris/esp/esp-at/components/at/lib\libesp8266_at_core.a(at_port.o):(.literal.esp_at_module_init+0x20): undefined reference to `xQueueCreateSet'
C:/Users/Chris/Documents/work/ESP/msys32/home/Chris/esp/esp-at/components/at/lib\libesp8266_at_core.a(at_port.o):(.literal.esp_at_module_init+0x24): undefined reference to `xQueueAddToSet'
C:/Users/Chris/Documents/work/ESP/msys32/home/Chris/esp/esp-at/components/at/lib\libesp8266_at_core.a(at_port.o):(.text.at_recv_task+0x4c): undefined reference to `xQueueSelectFromSet'
C:/Users/Chris/Documents/work/ESP/msys32/home/Chris/esp/esp-at/components/at/lib\libesp8266_at_core.a(at_port.o): In function `esp_at_module_init':
(.text.esp_at_module_init+0x74): undefined reference to `xQueueCreateSet'
C:/Users/Chris/Documents/work/ESP/msys32/home/Chris/esp/esp-at/components/at/lib\libesp8266_at_core.a(at_port.o): In function `esp_at_module_init':
(.text.esp_at_module_init+0x88): undefined reference to `xQueueAddToSet'
C:/Users/Chris/Documents/work/ESP/msys32/home/Chris/esp/esp-at/components/at/lib\libesp8266_at_core.a(at_port.o): In function `esp_at_module_init':
(.text.esp_at_module_init+0x98): undefined reference to `xQueueAddToSet'
collect2.exe: error: ld returned 1 exit status
make: *** [/home/Chris/esp/esp-at/esp-idf/make/project.mk:510：/home/Chris/esp/esp-at/build/esp-at.elf] 错误 1

xcguang, 2019-12-11:
是 ESP8266 吗？虚拟机？
遇到这种情况一般是 esp-idf 更新失败所致，导致 SDK 和 AT 工程不匹配，你可以看看编译开始的log；并确认 SDK 的版本是否有变动

hzm476172557, 2019-12-11:
> 是 ESP8266 吗？虚拟机？
> 遇到这种情况一般是 esp-idf 更新失败所致，导致 SDK 和 AT 工程不匹配，你可以看看编译开始的log；并确认 SDK 的版本是否有变动

是的，编译的是ESP8266的，不是虚拟机环境，按照文档搭建的msys32环境，esp-idf也是esp-at进行make defconfig时自动clone的。
然后后面步骤完全按照esp-at的指示操作，make没有异常，就是链接的时候报的。我现在准备吧idf删掉，重新make一次

xcguang, 2019-12-11:
进入 SDK，git status 看一下是否有改动；另外再 git log 看下，是不是和 https://github.com/espressif/esp-at/blob/master/module_config/module_esp8266_default/IDF_VERSION 里面的 commit id 匹配；或者进入到 SDK，手动切换到 https://github.com/espressif/esp-at/blob/master/module_config/module_esp8266_default/IDF_VERSION 所描述的 commit id

hzm476172557, 2019-12-11:
> 进入 SDK，git status 看一下是否有改动；另外再 git log 看下，是不是和 https://github.com/espressif/esp-at/blob/master/module_config/module_esp8266_default/IDF_VERSION 里面的 commit id 匹配；或者进入到 SDK，手动切换到 https://github.com/espressif/esp-at/blob/master/module_config/module_esp8266_default/IDF_VERSION 所描述的 commit id

Chris@DESKTOP-4G6U8QS MINGW32 ~/esp/esp-at
$ make flash
Makefile:36: There is no module_config/module_wroom-02,Using module_config/module_esp8266_default
/home/Chris/esp/esp-at/tools/mkfatfs/Makefile.projbuild:13: 警告：覆盖关于目标“clean”的配方
/home/Chris/esp/esp-at/esp-idf/components/app_update/Makefile.projbuild:56: 警告：忽略关于目标“clean”的旧配方
Toolchain path: /opt/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: crosstool-ng-1.22.0-100-ge567ec7b
Compiler version: 5.2.0
Python requirements from C:/Users/Chris/Documents/work/ESP/msys32/home/Chris/esp/esp-at/esp-idf/requirements.txt are satisfied.
Makefile:36: There is no module_config/module_wroom-02,Using module_config/module_esp8266_default
/home/Chris/esp/esp-at/tools/mkfatfs/Makefile.projbuild:13: 警告：覆盖关于目标“clean”的配方
/home/Chris/esp/esp-at/esp-idf/components/app_update/Makefile.projbuild:56: 警告：忽略关于目标“clean”的旧配方
CC build/bootloader/main/bootloader_start.o

以上是make flash 开始编译前的一段消息，看起来似乎也没有问题
我现在正在重新clone esp-idf。待会完了之后试试你说的方法~

hzm476172557, 2019-12-11:
> 进入 SDK，git status 看一下是否有改动；另外再 git log 看下，是不是和 https://github.com/espressif/esp-at/blob/master/module_config/module_esp8266_default/IDF_VERSION 里面的 commit id 匹配；或者进入到 SDK，手动切换到 https://github.com/espressif/esp-at/blob/master/module_config/module_esp8266_default/IDF_VERSION 所描述的 commit id

按你说的查了：
Chris@DESKTOP-4G6U8QS MINGW32 ~/esp/esp-at/esp-idf
$ git status
头指针分离于 aed79e96
无文件要提交，干净的工作区

Chris@DESKTOP-4G6U8QS MINGW32 ~/esp/esp-at/esp-idf
$ git log
commit aed79e96f18452ca34c8803a934b33ed9875effa (HEAD)
Merge: 22029d98 19529e4f
Author: Dong Heng <dongheng@espressif.com>
Date:   Tue Dec 3 11:33:05 2019 +0800

    Merge branch 'feature/esp-wolfssl' into 'master'

    esp-wolfssl: add esp-wolfssl component

    See merge request sdk/ESP8266_RTOS_SDK!1199

与https://github.com/espressif/esp-at/blob/master/module_config/module_esp8266_default/IDF_VERSION 里面的 commit id 匹配，但是编译依然相同的问题

hzm476172557, 2019-12-12:
已解决，打开FREERTOS QUEUE SETS功能即可

closed 2019-12-12 by hzm476172557

xcguang, 2019-12-12:
这个宏，在 Makefile 里面强制设置的，很奇怪你这编译不过，后面加入到 sdkconfig.defaults 里面去