espressif/esp-at #299: HTTP AT Command

LiThirteen, 2019-12-12:
问题1：使用HTTP AT Command，如何设置http连接超时的时长。
问题2：使用http client post请求后，接收不到服务器返回的分块编码的数据，只会返回ok，服务器返回的头中Transfer-Encoding: chunked， 当服务器返回的头中有Content-Length,才会显示数据+IPD:3,xx   OK，请问这个如何http at指令，如何控制能够读取服务器应答的分块编码的数据？

MrYanghaohaodsdasadasdasadasdas, 2019-12-12:
您好，目前`HTTP AT Command` 还在持续更新中，`HTTP AT Command `暂不支持` Header `字段的设置 以及  `chunked ` 数据的接收。

LiThirteen, 2019-12-12:
> 您好，目前`HTTP AT Command`还在持续更新中，`HTTP AT Command `暂不支持`Header`细分的设置以及 `chunked `数据的接收。
期待更新，那关于http连接超时的时长，也是不可以设置的吗，

xcguang, 2019-12-12:
你可以将你们当前的需求，都罗列出来，我们评估一下可行性

LiThirteen, 2019-12-12:
将数据实时传输到服务器，间隔1s，超时等待时间过长，想设置短一点，还想问一下，这个http AT 指令中的post请求方法，是长连接还是短连接？

MrYanghaohaodsdasadasdasadasdas, 2019-12-19:
请求 header 字段中未设置 keep-alive

LiThirteen, 2019-12-19:
@MrYanghaohaodsdasadasdasadasdas 感谢提醒，我去试试

xcguang, 2020-03-15:
问题怎样了？解决了吗？

LiThirteen, 2020-03-16:
@xcguang 解决了，更新了ESP8266-RTOS-SDK 、ESP-AT后解决了该问题。

closed 2020-03-16 by xcguang