espressif/esp-at #302: Incoming +IPD message during CIPSEND hangs ESP receiver

WavrSamuel, 2019-12-12:
We're using ESP-IDF v3.3 and AT library 2.0.0 in a ESP32-WROOM-32U connected to an external MCU. The issue is that when we're sending data, and an incoming +IPD arrives when the ESP32 is waiting for data from the MCU, the ESP32 never sends any more IPD messages. We're also tried this with various custom IDF builds and different baud rates, the result is always the same. 

```
// MCU request to send 1363 bytes
OK // ESP Responds 
> // ESP Waits for data, MCU sends 1363 bytes
Recv 1363 bytes // ESP responds 5 bytes received
SEND OK // ESP responds SEND OK
// MCU request to send 1363 bytes
OK // ESP Responds 
> // ESP Waits for data, MCU sends 1363 bytes
// "Bad arrival" of IPD .....
+IPD,4,5,"192.168.1.48",65267:<data> // ESP notifies MCU 5 bytes on conn 4
Recv 1363 bytes // ESP reponds 1363 bytes
SEND OK // ESP responds SEND OK
```

When the IPD message arrives after the MCU requests to send X bytes, and before the ESP has received X bytes, the ESP32 stops sending IPD messages for incoming packets. We can keep sending data, and the ESP32 keeps responding for all commands.

If the IPD message arrives after SEND OK and before our next CIPSEND, it works just fine, up until the above issue happens.

xcguang, 2019-12-12:
Thanks for your report, I think it has been fixed in commit ec7a3f2, and improvement process in commit 7f9307b. If possible, please try it again using the latest master branch.

WavrSamuel, 2019-12-12:
Thanks for the quick response! Tested, can confirm that it's gone in current master 7e33e61c

closed 2019-12-12 by WavrSamuel

NunoHub, 2020-05-12:
Under the same conditions, with commint 7e33e61, we have a similiar problem. If we issue a CIPSEND while receiving a big IPD message, the ESP completly ignore it, and if we receive the '>' and then start receiving the IPD before send the bytes or at the middle of it, we never receive the "recv xx bytes" message. If we send the CIPSEND and no IPD arrives during the process we have no problems.