espressif/esp-at #308: "delete\r\n" in middle of +IPD data

mjessome, 2019-12-20:
Hello,
I'm having issues with raw TCP and SSL connections.
When receiving very large payloads, I am seeing a `delete\r\n` response wedged in the middle of my IPD data.

I am doing HTTP request over raw TCP connection to download a 6k file full of null bytes, and the response looks like this:

```
+IPD,201:
0000: 48 54 54 50 2f 31 2e 30 20 32 30 30 20 4f 4b 0d  | HTTP/1.0.200.OK.
0016: 0a 53 65 72 76 65 72 3a 20 53 69 6d 70 6c 65 48  | .Server:.SimpleH
0032: 54 54 50 2f 30 2e 36 20 50 79 74 68 6f 6e 2f 33  | TTP/0.6.Python/3
0048: 2e 37 2e 35 0d 0a 44 61 74 65 3a 20 46 72 69 2c  | .7.5..Date:.Fri,
0064: 20 32 30 20 44 65 63 20 32 30 31 39 20 31 39 3a  | .20.Dec.2019.19:
0080: 34 39 3a 30 30 20 47 4d 54 0d 0a 43 6f 6e 74 65  | 49:00.GMT..Conte
0096: 6e 74 2d 74 79 70 65 3a 20 61 70 70 6c 69 63 61  | nt-type:.applica
0112: 74 69 6f 6e 2f 6f 63 74 65 74 2d 73 74 72 65 61  | tion/octet-strea
0128: 6d 0d 0a 43 6f 6e 74 65 6e 74 2d 4c 65 6e 67 74  | m..Content-Lengt
0144: 68 3a 20 36 30 30 30 0d 0a 4c 61 73 74 2d 4d 6f  | h:.6000..Last-Mo
0160: 64 69 66 69 65 64 3a 20 46 72 69 2c 20 32 30 20  | dified:.Fri,.20.
0176: 44 65 63 20 32 30 31 39 20 31 39 3a 34 38 3a 35  | Dec.2019.19:48:5
0192: 30 20 47 4d 54 0d 0a 0d 0a  | 0.GMT.....
+IPD,2920:
0000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0016: 00 00 00 00 00 00 64 65 6c 65 74 65 0d 0a 00 00  | ......delete....
0032: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0048: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0064: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0096: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0112: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0128: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0144: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0160: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0176: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0192: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0208: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0224: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0240: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0256: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0272: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0288: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0304: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0320: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
..............
```
Note the `delete\r\n` that is sent down in the middle of the `+IPD` response data. This is not a part of the payload and seems to be sent asynchronously when the other end of the connection hangs up.

I have tried lowering verbosity in `make menuconfig` at `Component Config > Log Output > Default log verbosity`, and it has not helped.

Any help with this would be greatly appreciated, thank you!

tvandergeer, 2019-12-31:
Can't provide you with a solution but I can confirm that I've seen this happen during my tests as well.

xcguang, 2020-01-09:
Did you change the log pin same as AT port pin?
It looks the log(`delete\r\n`) inserts into the data.
You can try these:
- `rm sdkconfig`, then `make SILENCE=1`

or 

- set the log pin different from AT port pin

xcguang, 2020-02-07:
Hi,
Is the issue solved? Can I close it?

xcguang, 2020-02-13:
Any update?  I will close this issue, you can reopen it if you have any question

closed 2020-02-13 by xcguang