espressif/esp-at #309: where to donwload AT Firmware 2.0 for ESP32 (but with included MQTT)

valoni, 2019-12-24:
i downloaded ready AT Firmware for ESP32 v2.0 but there not included MQTT commands.
where i can download variant that include MQTT as part of AT Commands.

regards,
VALONI

ustccw, 2019-12-25:
@valoni The released AT firmware does not contain MQTT features.
user can use MQTT by compiling AT SDK, following by [README.md](https://github.com/espressif/esp-at/blob/master/README.md)

ustccw, 2019-12-25:
If you're not interested in compiling。You could provide me your module type, I will provide a AT fireware according your module, here.

valoni, 2019-12-25:
i have an ESP32s module (ESP32-WROOM-32 [with default 4 Mb flash , 540 kb sram])
or exactly ESP32-DevKitC

kurta999, 2019-12-25:
Im interested too. Anyway consider release them with MQTT and BT, or create another releases where those are enabled by default. That would help for plenty of people, including me. Thanks.

ustccw, 2019-12-26:
@valoni Please download the bin from address 0x0, it will work on ESP32-WROOM-32.
[target.zip](https://github.com/espressif/esp-at/files/4001174/target.zip)

ustccw, 2019-12-26:
@valoni @kurta999 As the issues 284 said, https://github.com/espressif/esp-at/issues/284, We have a guidance about AT connect to AWS, but it is written in Chinese.
You also could take a look at [AT commands](https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md)

In fact, MQTT based on TLS depend on certificates, So it is recommended that you compile AT firmware according to [README.md](https://github.com/espressif/esp-at)

valoni, 2019-12-26:
> Im interested too. Anyway consider release them with MQTT and BT, or create another releases where those are enabled by default. That would help for plenty of people, including me. Thanks.

I'm too

valoni, 2019-12-26:
> @valoni @kurta999 As the issues 284 said, #284, We have a guidance about AT connect to AWS, but it is written in Chinese.
> You also could take a look at [AT [...]

can i use MQTT Based on TLS with this bin file that you provided to me

ustccw, 2019-12-26:
> > @valoni @kurta999 As the issues 284 said, #284, We have a guidance about AT connect to AWS, but it is written in Chinese.
> > You also could take a look at [AT commands](https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md)
> > In fact, MQTT based on TLS depend on certificates, So it is recommended that you compile AT firmware according to [README.md](https://github.com/espressif/esp-at)
> 
> @ustccw thank you for compile bin file
> butt can i use MQTT Based on TLS with this bin file that you provided to us

If there is no a certificates auth, it can works on TLS, otherwise, user has to compile the AT firmware.

valoni, 2020-01-09:
now i was able to compile via ubuntu 1806 ova (with vmware) ,
thanks a lot for help .

closed 2020-01-09 by valoni