espressif/esp-at #310: Use AT+CIPSEND from AT firmware

kzoltaan, 2019-12-28:
Hi,
Somewhere I saw, there is a esp_at_xxx command to use internal AT commands with parameters from lib_at. I don't find it. Do you know?

I want the following:
make SSL connection with host mcu with AT+CIPSTART
I get the linkID, and send back to esp with custom AT command
I dont have enough memory for "huge" JSON data in host mcu, I want to assemble this with esp32 and send with internal "+CIPSEND" command. I want to trigger this sending with custom AT command. Or how can I use this socket?

I want to try it. I wrote tls1.2 tcp client with custom AT command with auto data send, but I unable to reconnect to server when connection lost. something going wrong, ssl always takes error on reconnect... I try to find other method. 
Or do u have working TLS tcp client code with reconnect? :)

Thanks.

kzoltaan, 2019-12-29:
TLS tcp client working very well, so only the first question is relevant.... If we want to use built in feature...

xcguang, 2020-01-09:
1. Or how can I use this socket? 
 -- You can try `esp_at_custom_net_ops_regist`
2. Or do u have working TLS tcp client code with reconnect?
 -- Currently, you need re-create connection via MCU.

closed 2020-03-12 by xcguang