espressif/esp-at #311: Build issue when set UART 0 as both log pin and AT port pin

shahulakthar, 2019-12-30:
I try to build a modified bin which set UART 0 as both log pin and AT port pin.

I followed the procedure mentioned on [here](https://github.com/espressif/esp-at/blob/master/docs/How_To_Set_AT_Port_Pin.md)

I got the following error. Using windows 7. I don't know how to fix them. 

[image]

Module used - ESP WROOM 32
Any solution to this issue or any precompiled binary that will set UART 0 as both log pin and AT port pin will be great help.

Thanks.

shahulakthar, 2019-12-30:
> Generating esp32.project.ld
> Traceback (most recent call last):
>   File "C:/SPB_Data/esp/esp-idf/esp-at/esp-idf/tools/  ldgen/ldgen.py", line 113, in <module>
>     main()
>   File "C:/SPB_Data/esp/esp-idf/esp-at/esp-idf/tools/  ldgen/ldgen.py", line 96, in main
>     sdkconfig = SDKConfig(kconfig_file, config_file,   args.env)
>   File "C:/SPB_Data/esp/esp-idf/esp-at/esp-idf/tools/  ldgen/sdkconfig.py", line 56, in __init__
>     self.config = kconfiglib.Kconfig(kconfig_file.nam  e)
>   File "C:/SPB_Data/esp/esp-idf/esp-at/esp-idf/tools/  kconfig_new/kconfiglib.py", line 656, in __init__
>     self.top_node)  # prev_node
>   File "C:/SPB_Data/esp/esp-idf/esp-at/esp-idf/tools/  kconfig_new/kconfiglib.py", line 1583, in _parse_bloc  k
>     self._enter_file(sourced_file)
>   File "C:/SPB_Data/esp/esp-idf/esp-at/esp-idf/tools/  kconfig_new/kconfiglib.py", line 1167, in _enter_file
>     .format(self._filename, self._linenr, e.message))
> IOError: C:/SPB_Data/esp/esp-idf/esp-at/esp-idf/Kconf  ig:59: Could not open 'C:SPB_Dataespesp-idfesp-atesp-  idfcomponentsapp_updateKconfig.projbuild' (ENOENT: No   such file or directory). Perhaps the $srctree enviro  nment variable (which was unset) is set incorrectly.   Note that the current value of $srctree is saved when   the Kconfig instance is created (for consistency and   to cleanly separate instances). Also note that e.g.   $FOO in a 'source' statement does not refer to the en  vironment variable FOO, but rather to the Kconfig Sym  bol FOO (which would commonly have 'option env="FOO"'   in its definition).
> make: *** [/c/SPB_Data/esp/esp-idf/esp-at/esp-idf/com  ponents/esp32/Makefile.projbuild:51: /c/SPB_Data/esp/  esp-idf/esp-at/build/esp32/esp32.project.ld] Error 1
>

xcguang, 2020-02-07:
Please try it using the latest master branch

xcguang, 2020-03-04:
I will close the issue, if the problem persists, please reopen it.

closed 2020-03-04 by xcguang