espressif/esp-at #312: ESP8266 AT+CWLIF出现错误

LiThirteen, 2019-12-31:
现象：
ready
AT+CWMODE?
+CWMODE:2

OK
AT+CWLIF
+CWLIF:0.0.0.0,00:00:00:00:00:00
+CWLIF:0.0.0.0,00:00:00:00:00:00
+CWLIF:0.0.0.0,00:00:00:00:00:00
+CWLIF:0.0.0.0,00:00:00:00:00:00
+CWLIF:0.0.0.0,00:00:00:00:00:00
+CWLIF:0.0.0.0,00:00:00:00:00:00
+CWLIF:0.0.0.0,00:00:00:00:00:00
+CWLIF:0.0.0.0,00:00:00:00:00:00
+CWLIF:0.0.0.0,00:00:00:00:00:00
+CWLIF:64.0.80.0,50:7e:10:40:03:00
+CWLIF:144.191.16.64,40:e3:00:00:84:be
+CWLIF:52.52.46.49,2b:43:57:4c:49:46
+CWLIF:98.58.52.51,2e:34:36:2e:34:39
+CWLIF:52.58.51.57,3a:33:36:3a:32:65
+CWLIF:53.53.44.102,0d:0a:00:0a:00:35

这是什么情况，是需要更改什么配置嘛

xcguang, 2020-01-09:
你是哪个版本？操作步骤是怎样的？我们复现不了

LiThirteen, 2020-01-09:
@xcguang

LiThirteen, 2020-01-09:
@xcguang   这是下载官方demo编译后的固件，没做代码和menuconfig中的其他改动，运行后就出现了这种情况，以下是调试的信息，版本也打出来了

ready
AT+CWMODE?
+CWMODE:2

OK
AT+GMR
AT version:2.1.0.0-dev(f7f172e - Dec  4 2019 08:43:54)
SDK version:v3.2-291-gaed79e96-dirty
compile time(4850d05):Dec 16 2019 13:14:25
Bin version:2.0.0(Unknown)

OK
AT+CWLIF
+CWLIF:0.0.0.0,00:00:00:00:00:00
+CWLIF:0.0.0.0,00:00:00:00:00:00
+CWLIF:0.0.0.0,00:00:00:00:00:00
+CWLIF:0.0.0.0,00:00:00:00:00:00

LiThirteen, 2020-01-16:
@xcguang ,
感谢您能够测试并回答问题，该bin版本：log 使用uart1输出，uart0使用了交换引脚功能

[AT2.1.zip](https://github.com/espressif/esp-at/files/4070308/AT2.1.zip)

xcguang, 2020-01-17:
你这个版本确实能复现，我们用最新的版本测试复现不了，能切换到我们最新的版本吗？

xcguang, 2020-01-17:
你是不是改了 SDK 的东西了？

LiThirteen, 2020-01-17:
@xcguang  不排除修改SDK导致错误的可能性，我会去尝试使用最新的版本来解决这个问题。

LiThirteen, 2020-02-05:
@xcguang  请问是更新SDK版本到最新还是esp-at这个项目

xcguang, 2020-02-05:
对的，每次都是更新到最新的 esp-at 工程

xcguang, 2020-02-05:
更新之后，最好删除一下 sdkconfig，使用最新的工程，重新生成一下

xcguang, 2020-02-10:
怎样？这个问题不修改 sdk 还存在吗？

xcguang, 2020-03-02:
这个问题不修改 SDK 还有复现么？

LiThirteen, 2020-03-02:
@xcguang  您好，通过更新ESP8266 SDK、esp-at项目，已经解决了这个问题，现已正常使用该指令。

closed 2020-03-02 by LiThirteen