espressif/esp-at #313: Prevent AT configuration from being stored in flash NVS?

ikalogic, 2020-01-02:
Hi,

I'm using the AT commands to configure the ESP32 using an MCU.

I'm wondering is there is a way of preventing commands like "AT+CWJAP" from saving its configuration to NVS?

For example, I don't want my ESP32 wasting time at startup trying to connect to the last network, as in my application, I know this network may not be valid. 

I'm also worried about Flash wear, as I'll have to change configuration often.

Did I miss something in the AT commands ? Is there a way to do that? 

Do I have to re-compile and re-flash the AT commands for that to work?

Thanks!

xcguang, 2020-01-03:
Maybe https://github.com/espressif/esp-at/issues/296 can help you.

xcguang, 2020-03-12:
We have added AT+SYSSTORE, https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Commands_Set.md#Basic-AT, maybe it can match your requirement.

closed 2020-03-12 by xcguang