espressif/esp-at #315: AT+CIPSNTPTIME?查询到的时间总是1970年？

fsfxmfy, 2020-01-03:
按照https://blog.csdn.net/espressif/article/details/101714951所写步骤，前面3步都正常AT+CWMODE=1
回复：
OK
AT+CWJAP="ssid","password"
回复：
WIFI DISCONNECT
WIFI CONNECTED
WIFI GOT IP
OK
AT+CIPSNTPCFG=1,8,"ntp1.aliyun.com"
回复：
OK
AT+CIPSNTPTIME?
回复：
+CIPSNTPTIME:Thu Jan  1 00:02:08 1970
OK
不知原因何在？

AT+GMR
AT version:2.1.0.0-dev(044790d - Dec 30 2019 10:16:10)
SDK version:v3.3.1
compile time(d03b392):Jan  3 2020 15:41:07
Bin version:2.0.0(WROOM-32)

OK

ustccw, 2020-01-03:
@fsfxmfy 
Please make sure that your router device connectd can visit the internet.  
Moreover, could you please try with others sntp server.
```
AT+CIPSNTPCFG=1,8,"cn.pool.ntp.org"
```

MaJerle, 2020-01-03:
Catch is that you have to wait for some time before SNTPTIME responds with correct time.
Try to wait ~5 seconds between setting server and acquiring time.

fsfxmfy, 2020-01-03:
确实是网络问题，换了路由器就正常了。

closed 2020-01-03 by fsfxmfy