espressif/esp-at #316: AT+MQTTCONN 引起esp32复位

fsfxmfy, 2020-01-03:
按照https://blog.csdn.net/espressif/article/details/101714951流程，当执行AT+MQTTCONN=0,"a1bxxxxxxc-ats.iot.us-east-2.amazonaws.com",8883,1时会回复：
ready
WIFI CONNECTED
WIFI GOT IP
查看LOG:
all parameters are ready

cert len:1220
key len:1675
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400014dc  PS      : 0x00060230  A0      : 0x80001459  A1      : 0x3ffd7f90  
A2      : 0x000022b3  A3      : 0x000022af  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800944be  A9      : 0x3ffd7f60  
A10     : 0x3ffd8164  A11     : 0x00000001  A12     : 0x00060220  A13     : 0x00000003  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000022b3  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

ELF file SHA256: 5c32741dca854b86c3a5197c5029a2d6db3129a3779b507df9673f345049954d

Backtrace: 0x400014dc:0x3ffd7f90 0x40001456:0x3ffd7fa0 0x40001445:0x3ffd7fc0 0x401b5d61:0x3ffd7fe0 0x401b631c:0x3ffd8010 0x40121d5a:0x3ffd8040 0x401191c1:0x3ffd8070 0x40123d56:0x3ffd8090 0x40095731:0x3ffd80d0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6800
ho 0 tail 12 room 4
load:0x40078000,len:11780
load:0x40080400,len:6620
entry 0x40080774
[0;32mI (75) boot: Chip Revision: 1[0m
[0;32mI (75) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (41) boot: ESP-IDF v3.3.1 2nd stage bootloader[0m
[0;32mI (41) boot: compile time 05:37:38[0m
[0;32mI (41) boot: Enabling RNG early entropy source...[0m
[0;32mI (46) boot: SPI Speed      : 40MHz[0m
[0;32mI (50) boot: SPI Mode       : DIO[0m
[0;32mI (54) boot: SPI Flash Size : 4MB[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (62) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (69) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (77) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (84) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (92) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (99) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (106) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (114) boot: End of partition table[0m
[0;32mI (118) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (126) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x294cc (169164) map[0m
[0;32mI (194) esp_image: segment 1: paddr=0x001294f4 vaddr=0x3ffbdb60 size=0x03450 ( 13392) load[0m
[0;32mI (199) esp_image: segment 2: paddr=0x0012c94c vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (201) esp_image: segment 3: paddr=0x0012cd54 vaddr=0x40080400 size=0x032bc ( 12988) load[0m
[0;32mI (215) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0xfcdbc (1035708) map[0m
[0;32mI (582) esp_image: segment 5: paddr=0x0022cddc vaddr=0x400836bc size=0x1472c ( 83756) load[0m
[0;32mI (617) esp_image: segment 6: paddr=0x00241510 vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (631) boot: Loaded app from partition at offset 0x100000[0m
[0;32mI (631) boot: Disabling RNG early entropy source...[0m
2.0.0

max tx power=78,ret=0

esp32模块复位了？不知是何原因。

ustccw, 2020-01-09:
@fsfxmfy this issue has already fixed, please try with the latest version.

xcguang, 2020-01-09:
可以用 a5a8c741ffa0ad504321d078a2e7d70b6cd47afb 试

xcguang, 2020-01-09:
If the issue persists, please reopen it.

closed 2020-01-09 by xcguang

fsfxmfy, 2020-01-09:
It's ok, it has been fixed!