espressif/esp-at #319: Compiling AT failed - IOError: C:/msys32/home/Ati/esp/esp-idf/Kconfig:59. Could anybody send me binaries with BT Classic?

kurta999, 2020-01-08:
I tried to compile it myself and this is the result.

https://www.esp32.com/viewtopic.php?f=13&t=13777

Before this is fixed, could anybody send me pre-compiled AT with Bluetooth Classic + MQTT enabled? My priority is on BT Classic, if MQTT not included it doesn't matter.

xcguang, 2020-01-09:
[artifacts-31.zip](https://github.com/espressif/esp-at/files/4038297/artifacts-31.zip)
You can try this.

kurta999, 2020-01-09:
Thank you for helping me out.
Does this release requires WROVER, classic BT only works with WROVER?

Since log says:  <ESC>[0;31mE (750) cpu_start: Failed to init external RAM!<ESC>[abort() was called at PC 0x40081437 on core 0

xcguang, 2020-01-10:
Could you try compile firmware using the latest master branch? Before compiling, you'd better remove sdkconfig file, and then `make`

kurta999, 2020-01-11:
You also mean latest for ESP-IDF or I should use 3.3 as before?

xcguang, 2020-01-13:
Hi,
Yes, the old branch is based on IDF v3.3, and there is a compiling issue you mentioned.
Please use the latest master branch, and make sure you delete sdkconfig file in the project before you compile it. Or you can enable `CONFIG_SPI_FLASH_USE_LEGACY_IMPL` in menuconfig by yourself

kurta999, 2020-01-15:
Thanks!

Now I sucessfully compiled it on Ubuntu, on windows I was unable to do it.

kurta999, 2020-01-15:
Now it says that partition is too big for ota_0. But If I use your build, it accepts at firmware but this is the output. I would be very glad if it was working to replace our laird BT chips to ESP in future.

https://gtfkephost.hu/files/2020/03/15/lkzem1yi76.png

xcguang, 2020-01-16:
How do you download the firmware? Can you give a screenshot?

kurta999, 2020-01-16:
https://gtfkephost.hu/files/2020/03/16/e1xs1i2oi9.png

kurta999, 2020-01-23:
Any ideas please?

Emiel84, 2020-01-24:
Seems that enabling BT classic is not as simple as following the online instructions.
I'm trying to get this to work and spend many days already on this aswell.
Why was there not decided to make a release including BT classic support ?

xcguang, 2020-03-15:
May I close this issue？

kurta999, 2020-03-15:
The main issue is solved but BT still doesn't work so can be closed. It is possible please create another release with BT and all functions enabled for ESP Wrover in the future please? Thanks.

xcguang, 2020-03-16:
In the next release, we will enable it for WROVER32 module.

closed 2020-03-16 by xcguang