espressif/esp-at #320: ESP AT is not true response

hiflord, 2020-01-09:
HELLO. I have a problem with at command. I couldnt do correctly ESP32 AT firmware. After firmware adding finish I connected ttl converter to esp's 16th and 17th pins. What I transmit that is coming same.
[image]
[image]
Can  you help me pls. My device is ESP32 Devkitc V4.

hiflord, 2020-01-09:
Also I see 'ready' when I push en button. 
[image]

xcguang, 2020-01-09:
[QCOM.zip](https://github.com/espressif/esp-at/files/4040056/QCOM.zip)
Could you try this tool?

hiflord, 2020-01-09:
[image]
[image]
Same. It is returning What I send.

hiflord, 2020-01-09:
When I do send with enter It is done. Nearly 1 month I am working on it. It is need <CR> end of command. 
[image]
Thank you so much! Have a better life

xcguang, 2020-01-10:
Please choose this
[image]

closed 2020-01-10 by xcguang

hiflord, 2020-01-10:
When I send at commands, the same command coming first. After that at command answer is coming. Is it normal?
[image]
Thesee all responses of wifi module. I send data AT command from nucleo-l073rz and I am watching with ttl's receiver root.