espressif/esp-at #321: ESP AT Compiling

valoni, 2020-01-09:
tried to compile AT Firmware with Ethernet enabled (but seems is not included in latest sources on ESP_IDF)

ESP-Coco, 2020-01-13:
Hi @valoni , in IDF release/v4.0, the ethernet interface has changed. Now I will re-implement the Ethernet interface, and I will inform you when it is finished. Thank you for your attention.

ubichsel, 2020-01-15:
I'm looking forward for the Ethernet re-implementation as well. Thank you

ESP-Coco, 2020-02-28:
Hi @valoni Ethernet has been supported on the latest version( https://github.com/espressif/esp-at/commit/a577c7967bc92820ca34fe0f7df3760b75d08f4b ), you can pull it to test. Note that we currently support IP101 PHY by default. TLK110 PHY is no longer supported because TI stoped production.

valoni, 2020-02-28:
> Hi @valoni Ethernet has been supported on the latest version( [a577c79](https://github.com/espressif/esp-at/commit/a577c7967bc92820ca34fe0f7df3760b75d08f4b) ), you can pull it to test. Note that we currently support IP101 PHY by default. TLK110 PHY is no longer supported because TI stoped production.

why do not include also "ENC28J60 , Wiz5500" both are SPI and are most used ethernet modules ..

but let go with compile :)

ESP-Coco, 2020-02-28:
Hi @valoni ,I'm really sorry that not add SPI PHY support, because I don't have an SPI PHY device at hand to test. You can add SPI Ethernet support by comparing my modifications and [Ethernet demo](https://github.com/espressif/esp-idf/tree/release/v4.0/examples/ethernet/basic). I look forward to receiving your feedback

ESP-Coco, 2020-03-02:
Hi @valoni， The different hardware needs different configuration, I don't know what hardware configuration you are using now, you can refer to this [ducument](https://github.com/espressif/esp-idf/tree/master/examples/ethernet#common-configurations) ,If problems still occur, could you please send us your hardware schematic?

valoni, 2020-03-02:
@ESP-Coco  

I'm abroad for test  - but for test i am using those hardware 
(since my communication was based on AT Firmware and WiFi was disabled and forbidden to use for our clients)

but those days i'll return and will check and test  boards i have 

**Esp32 DevKit (small board)**
https://www.amazon.com/Espressif-Genuine-ESP32-DEVKITC-ESP-WROOM-32-soldered/dp/B01N0SB08Q 

**DoIt Esp32 Devkit 1** 
https://github.com/playelek/pinout-doit-32devkitv1

and this 
**D1 - Mini ESP32** - it fullfill our needs too
https://www.banggood.com/Geekcreit-D1-Mini-ESP32-ESP-32-WiFibluetooth-Internet-Of-Things-Development-Board-Based-ESP8266-p-1205854.html?cur_warehouse=CN

**for Ethernet I use :**
Waveshare LAN PHY8270
https://www.amazon.com/Waveshare-performance-LAN8720-Board-Transceiver/dp/B00N4JW3YK

**noname LAN PHY8720**
https://www.ebay.com/itm/1pcs-Smart-Electronics-LAN8720-network-module-Ethernet-transceiver-for-arduino/183479331440

and
**ENC26J80 - Eth Click Mikroe (very high quality board)**
https://www.mikroe.com/eth-click

**noname - ENC26J80  (work well on arduino and STM32) - it fullfill our needs too**
https://www.ebay.com/itm/New-ENC28J60-Ethernet-LAN-Network-Module-For-Arduino-SPI-AVR-PIC-LPC-STM32/141976891928

also 
Eth Wiz W5500 Click Board (very quality board)
https://www.mikroe.com/eth-wiz-click

and 
no name - W5500 (work well on arduino and STM32)
https://www.ebay.com/itm/W5500-Ethernet-Network-Module-TCP-IP-51-STM32-SPI-Interface-For-Arduino/392301629351 

p.s.
seems actual ova have breaks on compiler and do not compile latest AT firmware 
http://download.espressif.com/esp_share/env/ubuntu1804.ova  and need to be updated 

regards,
VALONI

ESP-Coco, 2020-03-02:
Hi @valoni  ENC28J60 is very old, only 10Mbps Ethernet. and it just supported in our latest IDF master branch. You can test it use [enc28j60 demo](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/enc28j60)
W5500 goes through the hardware protocol stack, but we use the lwip software protocol, so it is not supported.
Now back to your question, we want to know the wiring between your ESP32 and the PHY.  I think you can refer to our[ Ethernet development board description](https://docs.espressif.com/projects/esp-idf/en/latest/hw-reference/get-started-ethernet-kit.html), and the corresponding [hardware schematic](https://dl.espressif.com/dl/schematics/SCH_ESP32-ETHERNET-KIT_A_V1.1_20190711.pdf).

valoni, 2020-03-02:
> Hi @valoni ENC28J60 is very old, only 10Mbps Ethernet. and it just supported in our latest IDF master branch. You can test it use [enc28j60 [...]

did it was supported on AT or as stand alone application (i want to include with AT Firmware - ENC26J80) 

i do not know how to setup AT Firmware Ethernet Enabled (with ENC26J80) ????
[image]

can you help me how to include (or what to setup to make AT Firmware with Ethernet Enabled - ENC26J80)

ESP-Coco, 2020-03-12:
Hi @valoni ENC28J60 is supported on the master develop branch, and AT currently only supports IDF release/v4.0. In addition, I don't have this Ethernet in my hand, so I can't test it. You can include it by yourself.
1. Fristly, you need to use the ENC28J60 demo ( https://github.com/espressif/esp-idf/tree/master/examples/ethernet/enc28j60#enc28j60-example ) to test whether you can get the IP. This is very helpful in determining whether there is a problem with the hardware. 
2. After you've done this step, porting the demo to release/v4.0 to run on, you need to alse can get the IP.
3. Just copy the ENC28J60 init code to  at_eth_init.c. refer to our demo (https://github.com/espressif/esp-idf/blob/release/v4.0/examples/ethernet/basic/main/ethernet_example_main.c#L92)

closed 2020-08-18 by xcguang