espressif/esp-at #322: AT+CIPSTATUS returns "ERROR"

carlessole, 2020-01-09:
Hello,
In some cases after a heavy TCP communication (receiving it via +IPD) the ESP32 stops working OK and the TCP communication is now working anymore.
When this happens, if I ask "AT+CIPSTATUS" I get this answer from ESP32:
[image]

Do you know what can cause that? I'm not doing anything strange apart from the heavy TCP communication.

I'm using last esp32 release: [v2.0.0_esp32](https://github.com/espressif/esp-at/releases/tag/v2.0.0.0_esp32)

Thanks!

SUNLUXIN, 2020-01-20:
Can you describe your case in detail and provide some log and the commands you send?

mihaiberevoescu, 2020-01-21:
Hi! I also have this problem.
I have an ESP32 with AT command v2.0.0.0. Problem appear at random point, but when it appear is consistent. ESP is controlled by a STM32 host and last commands given to esp32 before problem appear are:
ESP32 cmd: +PING="www.rotld.ro"
ESP32 response 13 B: +PING:258.... [trailing dots are \r\n\r\n]
ESP32 cmd: +CIPSTATUS
ESP32 response 61 B: STATUS:3..+CIPSTATUS:0,"TCP","192.168.123.10",443,57363,0....
ESP32 cmd: +CWJAP?
ESP32 response 52 B: +CWJAP:"myaccp","02:0c:e7:43:47:57",11,-37,0,0,0....
ESP32 cmd: +CIPSEND=0,495
[now I'm sending 495 bytes from host to ESP32]
ESP32 cmd: +CWJAP?
ESP32 response 52 B: +CWJAP:"myaccp","02:0c:e7:43:47:57",11,-39,0,0,0....
ESP32 cmd: +CIPSTATUS
ESP32 responded with ERROR [entire ESP32 response is: "STATUS:3\r\n\r\nERROR\r\n"

According to documentation, STATUS:3 is "The ESP32 Station has created a TCP or UDP transmission" but no connection is shown and from this point down everything related to stream 0 fails. AT+CIPSEND return an error, AT+CIPSTATUS also return an error. So the reported status is correct, but stream is somehow missing.

xcguang, 2020-03-12:
Can you try it using the latest master branch? It seems we have fixed it.

mihaiberevoescu, 2020-03-12:
Hi Xcguang, thanks for update.
Unfortunately, is not easy for me to test because my esp32 module is soldered to a PCB and, despite I do have the possibility to connect to uart0 to manually load the bin, it is not as simple as AT+CIUPDATE. On the other hand, the problem does not appear each and every time so the subsequent testing should be long enough to confirm the bug is solved. When I will test and have an updated response, I will let you know.

xcguang, 2020-03-12:
Thanks, if you have any question, just feel free to let us know.

closed 2020-08-18 by xcguang