espressif/esp-at #323: [ESP8266] UDP not working in passive mode (CIPRECVMODE=1)

tsvehagen, 2020-01-10:
When using passive mode on the ESP8266 it is not possible to receive data over UDP. When data is received on a UDP link, the ESP8266 just send `+IPD,<linkid>,<len>` without any data.

See https://github.com/espressif/esp-at/issues/223#issuecomment-568238797

jandrassy, 2020-01-10:
it applies to ESP32 too

tsvehagen, 2020-01-10:
> it applies to ESP32 too

Updated title and comment :+1:

closed 2020-01-14 by igrr

jandrassy, 2020-01-17:
UDP is receive is passive in passive mode. nice. thank you

jandrassy, 2020-01-18:
only one  AT+CIPRECVDATA is possible. second returns ERROR even if there should be more data.

next message arriving before the previous message was read is concatenated to the data available 
for read. 

example:
host has a 256 bytes buffer
a 306 bytes long UDP message is announced with +IDP
host reads 256 bytes with  AT+CIPRECVDATA. 100 bytes are lost. it can't be read with additional  AT+CIPRECVDATA

another example
host has a 256 bytes buffer
two 153 bytes long UDP messages arrive and are announced with +IDP,x,306
host reads 256 bytes with  AT+CIPRECVDATA. the buffer is filled with data of both messages and 50 bytes of the second message are lost

jandrassy, 2020-01-19:
please allow to read out the UDP messages with multiple AT+CIPRECVDATA
```
+IPD,4,331
AT+CIPRECVDATA=4,256
+CIPRECVDATA,256,<256 bytes of data>
OK
AT+CIPRECVDATA=4,75
ERROR
```