espressif/esp-at #324: AT+BLEGATTCWR  how to send sequence of bytes as hex values

marcrupprath, 2020-01-11:
A typical command / value set i have to send to my peripheral charactersistics lokks like:
 uint8_t SendData[]={0x08,0x00,0x81,0x00,0x10,0x51,0x00,0xA0};

How to send the number of hex bytes using BLEGATTCWR write what would be the correct form
i have to type in when prompt appears  in terminal programm i use for test?

Thanks in advance

xiewenxiang, 2020-01-14:
All you need to do is specify a length of 8 bytes and then enter the data.

marcrupprath, 2020-01-19:
Hello,
i am sorry but that was not the question:
The question is:
After i specified lenghth , what is correct spelling at prompt.
Example: When i send "AAAAAAAA" i receive :4141414141414141 > HEX representation of my string send, thats ok but When i have to send hex values like 0x08,0x00,0x81,0x00,0x10,0x51,0x00,0xA0

How to type in at prompt so that they are reveived as : 0x08,0x00,0x81,0x00,0x10,0x51,0x00,0xA0

What must the input string  at the prompt look like, would you please give me a tested example on how to send fex values, thank you

xiewenxiang, 2020-03-12:
Hi,
Did you use the Serial port tools and input the command to test it? 
If so, you can do as this:

 - Select the input format of the serial tool as string.
 - Input command `AT+BLEGATTCWR=<conn_index>,<srv_index>,<char_index>[,<desc_index>],8`
 - then, you will get the response `>`
 - Select the input format of the serial tool as HEX, and input the hex value `08008100105100A0`

marcrupprath, 2020-03-13:
Thanks,
i will try if i ahev time.
meanwhile i changed concpt.

closed 2020-03-13 by marcrupprath