espressif/esp-at #325: Communication interrupt when +IPD occur in AT+CIPSEND process

fcl-a, 2020-01-16:
Firmware version: 
AT version:2.0.0.0(e44648c - Nov  4 2019 02:18:57)
SDK version:v3.3
compile time(7cac7ad):Nov  4 2019 06:11:40
Bin version:2.0.0(WROOM-32)

Test condiction:

ESP32 connecting to a TCP server which will send message "test x" around every 200ms. ESP32 will echo immediately when got message.

Issue occur condition:
Sometime the +IPD will occur between AT+CIPSEND command and "Recv x bytes", then ESP32 won't get the fellow messages from the server. After 2~3 min, the server will show the connection is closed but the ESP32 still show the connection is alive but will respone SEND FAIL to AT+CIPSEND 

[2020-01-14 12:05:09,091][DEBUG]:<COM40"192.168.4.2">[0:00:11.605]:Send:b'AT+CIPSEND=0,8'
[2020-01-14 12:05:09,099][DEBUG]:<COM40"192.168.4.2">[0:00:11.613]:Recv:b'AT+CIPSEND=0,8\r\n\r\nOK\r\n\r\n>'
[2020-01-14 12:05:09,100][DEBUG]:<COM40"192.168.4.2">[0:00:11.613]:Send:b'test,26,'
[2020-01-14 12:05:09,109][DEBUG]:<COM40"192.168.4.2">[0:00:11.623]:b'\r\nRecv 8 bytes\r\n\r\nSEND OK\r\n'
[2020-01-14 12:05:09,510][DEBUG]:<COM40"192.168.4.2">[0:00:12.024]:Recv:b'\r\n+IPD,0,8:test,28,\r\n'
[2020-01-14 12:05:09,510][DEBUG]:<COM40"192.168.4.2">[0:00:12.024]:got 1 msg, COM40"192.168.4.2":ipd=14, msg=14, retry=None,send=13
[2020-01-14 12:05:09,510][DEBUG]:<COM40"192.168.4.2">[0:00:12.024]:Send:b'AT+CIPSEND=0,8'
[2020-01-14 12:05:09,517][DEBUG]:<COM40"192.168.4.2">[0:00:12.031]:Recv:b'AT+CIPSEND=0,8\r\n\r\nOK\r\n\r\n>'
[2020-01-14 12:05:09,518][DEBUG]:<COM40"192.168.4.2">[0:00:12.032]:Send:b'test,28,'
[2020-01-14 12:05:09,526][DEBUG]:<COM40"192.168.4.2">[0:00:12.039]:**b'\r\n+IPD,0,8:test,30,\r\n\r\nRecv 8 bytes\r\n\r\nSEND OK\r\n**

iabdurrahman, 2020-05-29:
This bug also affect me, I write a simple server program to test this bug, you can follow the instruction in README.md to reproduce this bug

server program at [https://github.com/iabdurrahman/echo_server](https://github.com/iabdurrahman/echo_server)

vadimpanferoff, 2026-01-15:
The same bug.

denravonska, 2026-10-26:
> 6\. This is expected behavior.
>     The messages you see are URCs (Unsolicited Result Codes), which may be reported asynchronously during AT command execution.
>     The host MCU is expected to actively handle and parse these URCs.
>     The currently supported URCs are documented here:
>     https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/index.html#at-messages

That makes no sense. I'd argue that the engine should _never_ send URCs during AT command execution. The other way around makes it impossible to handle. For example, a firmware that handles "+IPD" as URC will contain the string "+IPD" which will then show up as a fake URC during a firmware or file transfer. How are you supposed to handle that?