espressif/esp-at #326: esp32_at固件menuconfig中使能BT-但是依旧无法使用BT功能

woshi520laoxie, 2020-01-16:
1、AT固件
2、muconfig中配置
 [ ] AT ble command support.                                                
   [ ] AT ble hid command support.  
   [*] AT bt command support.       
   [*]   AT bt spp command support.  
   [ ]   AT bt a2dp command support. 
   [*] AT FS command support. 

开启了 支持BT指令，取消了ble
但是实际使用过程中，依旧无法使用BT指令
如果需要使能BT指令是否还需要配置其他项目。有没有可以使用bt指令的**sdkconfig**
**sdkconfig**

woshi520laoxie, 2020-01-16:
好的：

发送数据 b'ATE0\r\n' 
 
接收:ATE0 OK 

发送数据 b'AT+BTINIT=1\r\n' 
 
接收:AT+BTINIT=1 ERROR 

 
发送数据 b'AT+BTSPPINIT=2\r\n' 
 
接收:AT+BTSPPINIT=2 ERROR 

 
发送数据 b'AT+BTNAME="XWZ-EXAMPLE"\r\n' 
 
接收:AT+BTNAME="XWZ-EXAMPLE" ERROR 

 
发送数据 b'AT+BTSCANMODE=2\r\n' 
 
接收:AT+BTSCANMODE=2 ERROR 

 
发送数据 b'AT+BTSECPARAM=3,1,"9527"\r\n' 
 
接收:AT+BTSECPARAM=3,1,"9527" ERROR 

 
发送数据 b'AT+BTSPPSTART\r\n' 
 
接收:AT+BTSPPSTART ERROR

可以提供下您下发命令log吗

xiewenxiang, 2020-01-16:
@woshi520laoxie 
编译这个固件的 sdkconfig文件还在吗？在的话 贴一下

xiewenxiang, 2020-01-16:
@woshi520laoxie 以及您编译出来的 esp_at.bin （只要这一个）

woshi520laoxie, 2020-01-16:
好的

woshi520laoxie, 2020-01-16:
你好啊，请问有查到原因吗？
你有没有可以使用BT的 sdkconfig呢？
我也试试看

------------------ 原始邮件 ------------------
发件人: "我自己的邮箱"<563376881@qq.com&gt;;
发送时间: 2020年1月16日(星期四) 下午4:03
收件人: "espressif/esp-at"<reply@reply.github.com&gt;;

主题: 回复： [espressif/esp-at] esp32_at固件menuconfig中使能BT-但是依旧无法使用BT功能 (#326)

好的

woshi520laoxie, 2020-01-16:
你这边可以也给我一个可以使用BT指令的sdkconfig吗

我没有收到你贴的sdkconfig和bin文件啊，我在等你发呢

woshi520laoxie, 2020-01-16:
不好意思啊，我一直以为发送过去了，估计刚才发的没有成功，但是我这边显示成功了

woshi520laoxie, 2020-01-16:
是否看得到附件，我这边发了第三次了，能否收到

------------------ 原始邮件 ------------------
发件人: "我自己的邮箱"<563376881@qq.com&gt;;
发送时间: 2020年1月16日(星期四) 下午5:40
收件人: "espressif/esp-at"<reply@reply.github.com&gt;;

主题: 回复： [espressif/esp-at] esp32_at固件menuconfig中使能BT-但是依旧无法使用BT功能 (#326)

你这边可以也给我一个可以使用BT指令的sdkconfig吗

xiewenxiang, 2020-01-16:
@woshi520laoxie 你直接发我邮箱看看

xiewenxiang@espressif.com

xiewenxiang, 2020-01-16:
@woshi520laoxie 

[temp.zip](https://github.com/espressif/esp-at/files/4070928/temp.zip)

woshi520laoxie, 2020-01-17:
你好啊，我试了一下你给的配置，的确是可以的
我想了解一下我的那个配置之所以不行的原因是什么呢？我记录下来，之后注意一下

woshi520laoxie, 2020-01-20:
你好啊，我测试过，你这边给我的配置文件是可以使用BT和BLE的
我想了解一下，我之前用的那个配置方式，是哪里出了问题吗？
我以后注意一下。

xcguang, 2020-02-08:
你可以对比一下配置文件

closed 2020-03-12 by xcguang