espressif/esp-at #327: esp_at编译出来esp-at.bin没有显示版本号

woshi520laoxie, 2020-01-17:
请问如何配置esp_at编译出来的bin文件可以显示版本号呢？
我在make menuconfig里面配置版本号
 2.0.0) AT firmware version.    
在IOT升级的时候无法解析到版本号，这个版本号的

2、我查看esp-at.bin里面的文件。每次编译出来都是
45 53 50 2d 41 54 20 20 20 20 20 20 20 20 20 20  
[image]
显示的都是如图显示

3、有没有办法可以修改esp_at.bin里面的版本号呢？我试过手动修改版本号是有效的，
     不过涉及到校验，校验的方式我目前还不知道。

4、能否告知如何写入版本号到esp_at.bin呢？

xcguang, 2020-01-17:
`
AT version:2.1.0.0-dev(f7f172e - Dec  4 2019 08:43:54)
SDK version:v3.2-291-gaed79e96
compile time(4850d05):Jan 17 2020 18:00:52
Bin version:2.0.0(Unknown)

OK
`
这里的 `Bin version` 吗？

woshi520laoxie, 2020-01-17:
不是的，是编译出来的esp_at.bin这个二进制文件
里面有个版本号。在做IOT升级的时候有一个判断，就是判断该版本和以前的版本的版本号是否一致
esp_at.bin最前面的数据中包含了一些有用的信息，如果IOT升级需要判断这个位置，那么这个位置应该有办法，通过
配置或者修改代码修改该版本。

xcguang, 2020-01-17:
你那个不是版本信息，是工程名称，你通过如下两种方法修改：
1. 修改根目录下 `export PROJECT_VER = "ESP-AT"`，然后 make 编译。后面我们准备删除这种方式，全部采用第二种方式
2. 更新到我们最新的版本，在根目录下放一个 version.txt 文件，通过 ./esp-idf/tools/idf.py build 编译，如下
[image]

woshi520laoxie, 2020-01-18:
你好啊，我不确定我目前用的是不是最新的esp_at，不过我想使用git拉取最新的代码

使用git pull

error: 您对下列文件的本地修改将被合并操作覆盖：
&nbsp;&nbsp;&nbsp; Makefile
&nbsp;&nbsp;&nbsp; components/at/lib/VERSION
&nbsp;&nbsp;&nbsp; components/at/lib/libesp32_at_core.a
&nbsp;&nbsp;&nbsp; components/at/lib/libesp32_at_core_silence.a
&nbsp;&nbsp;&nbsp; ...

&nbsp;&nbsp;&nbsp; ...

&nbsp;&nbsp;&nbsp; module_config/module_wrover-32/IDF_VERSION
&nbsp;&nbsp;&nbsp; module_config/module_wrover-32/sdkconfig.defaults
&nbsp;&nbsp;&nbsp; module_config/module_wrover-32/sdkconfig_silence.defaults
请在合并前提交或贮藏您的修改。
终止中

我看了一下是没有 CMakeLists.txt&nbsp; 这个文件的

所以我运行

./esp-idf/tools/idf.py build 的时候报如下异常

xwz@xwz-virtual-machine:/home/sambaShare/esp/code/esp-at$ ./esp-idf/tools/idf.py build
Setting IDF_PATH environment variable: /home/sambaShare/esp/code/esp-at/esp-idf
Checking Python dependencies...
Python requirements from /home/sambaShare/esp/code/esp-at/esp-idf/requirements.txt are satisfied.
CMakeLists.txt not found in project directory /home/sambaShare/esp/code/esp-at

2、我将 CMakeLists.txt 加入到工程中

再次进行编译&nbsp; ./esp-idf/tools/idf.py build

CMake Error at esp-idf/tools/cmake/project.cmake:95 (__project):
&nbsp; The CMAKE_ASM_COMPILER:

&nbsp;&nbsp;&nbsp; xtensa-esp32-elf-gcc

&nbsp; is not a full path and was not found in the PATH.

&nbsp; Tell CMake where to find the compiler by setting either the environment
&nbsp; variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
&nbsp; to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
&nbsp; CMakeLists.txt:88 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "/home/sambaShare/esp/code/esp-at/build/CMakeFiles/CMakeOutput.log".
See also "/home/sambaShare/esp/code/esp-at/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

那么你们是用什么指令更新最新的版本的呢？

woshi520laoxie, 2020-01-18:
你好，第一种方法，修改根目录下的什么文件呢？

woshi520laoxie, 2020-01-18:
你好啊，
1、因为编译目前我这边的esp_at使用的是网上下载编译好的交叉编译器。
Toolchain path: /home/sambaShare/esp/xtensa-esp32-elf/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
是主动叫该路径添加到PATH路径中

2、在esp_at路径下的esp-idf中有  CMakeLists.txt

3、由于教程中的idf教程用的是 新的，编译出来的版本是

gcc version 8.2.0 (crosstool-NG esp-2019r2)
该版本无法用于我目前的esp_at当中。

xcguang, 2020-01-19:
1. 修改的是 esp-at 根目录下的 Makefile
2. 请使用一个空的文件夹重新下载，不要指定 IDF-PATH
3. 最新的编译器可以从 https://docs.espressif.com/projects/esp-idf/en/v4.0-beta1/get-started-legacy/linux-setup.html 下载

woshi520laoxie, 2020-01-19:
你好：
       烧录代码报的这个异常是什么意思呢？
       idf.py build
       idf.py -p /dev/ttyUSB0 flash

--------------------------------------------------------》

esptool.py -p /dev/ttyUSB0 -b 460800 --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin 0x10000 recognition_wechat.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address&gt; <filename&gt; [<address&gt; <filename&gt; ...]
esptool write_flash: error: argument <address&gt; <filename&gt;: Detected overlap at address: 0x8000 for file: partition_table/partition-table.bin
esptool.py failed with exit code 2

 

 
 
最新的编译器可以从 https://docs.espressif.com/projects/esp-idf/en/v4.0-beta1/get-started-legacy/linux-setup.html 下载

xcguang, 2020-02-07:
看不到你提交完整信息

xcguang, 2020-03-06:
你这个问题怎样了？

woshi520laoxie, 2020-03-06:
你好啊。&nbsp; &nbsp; &nbsp;不好意思啊，谢谢你的帮助，这个问题已经解决了的。
&nbsp; &nbsp; &nbsp; 按照你说的方式，可以显示版本号了

closed 2020-03-06 by xcguang