espressif/esp-at #329: how to implement custom commands in AT

lakshmisivareddy, 2020-01-23:
Hi  i want to implement  my own custom AT command 
is there any example available ?

ubichsel, 2020-01-24:
I had problems finding documentation as well. In main you can call esp_at_custom_cmd_array_regist. Below a sample how I did the ALPN patch (in app_main.c):

This code before void app_main()
`char at_cust_alpnName[50];
static uint8_t at_setupCmdAlpn(uint8_t para_num)
{
	uint8_t* pAlpnName;
    if (para_num != 1) {
        return ESP_AT_RESULT_CODE_ERROR;
    }
    if (esp_at_get_para_as_str(0, &pAlpnName) != ESP_AT_PARA_PARSE_RESULT_OK) {
        return ESP_AT_RESULT_CODE_ERROR;
    }
    if (strlen((char*)pAlpnName) > (sizeof(at_cust_alpnName) - 1))
    {
    	return ESP_AT_RESULT_CODE_ERROR;
    }
    strcpy(at_cust_alpnName, (char*)pAlpnName);
    return ESP_AT_RESULT_CODE_OK;
}

uint8_t at_queryCmdAlpn(uint8_t *cmd_name)
{
	uint8_t buffer[70];
	snprintf((char*)buffer,sizeof(buffer) - 1,"%s:\"%s\"\r\n",cmd_name,at_cust_alpnName);
    esp_at_port_write_data(buffer,strlen((char*)buffer));
    return ESP_AT_RESULT_CODE_OK;
}

static esp_at_cmd_struct at_custom_cmdAlpn[] = {
		{"+CIPSSLALPN", NULL, at_queryCmdAlpn, at_setupCmdAlpn, NULL}
};`
And then register it on the very end of the file (before at_custom_init() in void app_main())
`esp_at_custom_cmd_array_regist (at_custom_cmdAlpn, sizeof(at_custom_cmdAlpn)/sizeof(at_custom_cmdAlpn[0]));`

Hope it helps you.

lakshmisivareddy, 2020-01-24:
Thanks for the response
i tried something  similar to this registering call backs to 'esp_at_cmd_struct' it worked for me 
Regards
LSREDDY

closed 2020-01-24 by lakshmisivareddy

espressifelse, 2020-06-02:
The structure for a custom command is outlined in esp_at.h

at command name
Test Command
Query Command
Setup Command
Execute Command

Here is a simple example with the same functionality as AT+RST

at_uart_task.c

`
static esp_at_cmd_struct at_custom_cmd[] = {
    {"+UART", NULL, at_queryCmdUart, at_setupCmdUartDef, NULL},
    {"+UART_CUR", NULL, at_queryCmdUart, at_setupCmdUart, NULL},
    {"+UART_DEF", NULL, at_queryCmdUartDef, at_setupCmdUartDef, NULL},
    {"+RESET", NULL, NULL, NULL, at_exeCmdReset},
};

static uint8_t at_exeCmdReset(uint8_t *cmd_name)//reset the chip
{
	esp_restart();
	for(;;){
	}
    return ESP_AT_RESULT_CODE_ERROR;
}
`