espressif/esp-at #330: Enable BT classic

Emiel84, 2020-01-23:
After following all the steps to compile the AT library and solving an issue in the ESP-IDF library :
https://github.com/espressif/esp-idf/issues/4658

I have been able to build the AT command set successfully.
But so far the ESP-WROOM-32 seems not be able to run anymore after a second try to get the AT command set to support bluetooth classic.

<ESC>[0;32mI (254) esp_image: segment 4: paddr=0x00140018 vaddr=0x400d0018 size=0x130d74 (1248628) map<ESC>[0m<CR><LF>
<ESC>[0;32mI (696) esp_image: segment 5: paddr=0x00270d94 vaddr=0x40083258 size=0x17d54 ( 97620) load<ESC>[0m<CR><LF>
<ESC>[0;32mI (737) esp_image: segment 6: paddr=0x00288af0 vaddr=0x400c0000 size=0x00064 (   100) load<ESC>[0m<CR><LF>
<ESC>[0;31mE (738) esp_image: Image length 1608576 doesn't fit in partition length 1572864<ESC>[0m<CR><LF>
<ESC>[0;31mE (743) boot: OTA app partition slot 0 is not bootable<ESC>[0m<CR><LF>
<ESC>[0;31mE (749) esp_image: image at 0x280000 has invalid magic byte<ESC>[0m<CR><LF>
<ESC>[0;31mE (755) boot: OTA app partition slot 1 is not bootable<ESC>[0m<CR><LF>
<ESC>[0;31mE (761) boot: No bootable app partitions in the partition table<ESC>[0m<CR><LF>
<ESC>[0;32mI (60) boot: ESP-IDF v3.3-dirty 2nd stage bootloader<ESC>[0m<CR><LF>
<ESC>[0;32mI (60) boot: compile time 16:16:47<ESC>[0m<CR><LF>
<ESC>[0;32mI (60) boot: Enabling RNG early entropy source...<ESC>[0m<CR><LF>
<ESC>[0;32mI (64) boot: SPI Speed      : 40MHz<ESC>[0m<CR><LF>
<ESC>[0;32mI (68) boot: SPI Mode       : DIO<ESC>[0m<CR><LF>
<ESC>[0;32mI (72) boot: SPI Flash Size : 4MB<ESC>[0m<CR><LF>
<ESC>[0;32mI (76) boot: Partition Table:<ESC>[0m<CR><LF>
<ESC>[0;32mI (80) boot: ## Label            Usage          Type ST Offset   Length<ESC>[0m<CR><LF>
<ESC>[0;32mI (87) boot:  0 phy_init         RF data          01 01 0000f000 00001000<ESC>[0m<CR><LF>
<ESC>[0;32mI (94) boot:  1 otadata          OTA data         01 00 00010000 00002000<ESC>[0m<CR><LF>
<ESC>[0;32mI (102) boot:  2 nvs              WiFi data        01 02 00012000 0000e000<ESC>[0m<CR><LF>
<ESC>[0;32mI (109) boot:  3 at_customize     unknown          40 00 00020000 000e0000<ESC>[0m<CR><LF>
<ESC>[0;32mI (117) boot:  4 ota_0            OTA app          00 10 00100000 00180000<ESC>[0m<CR><LF>
<ESC>[0;32mI (125) boot:  5 ota_1            OTA app          00 11 00280000 00180000<ESC>[0m<CR><LF>
<ESC>[0;32mI (132) boot: End of partition table<ESC>[0m<CR><LF>
<ESC>[0;32mI (136) boot: No factory image, trying OTA 0<ESC>[0m<CR><LF>

The Wroom was having ESP32-WROOM-32_AT_Bin_V2.0 installed before as a baseline.

I have been performing the following steps in the menuconfig for building the AT set.

Enable BT classic
Component config -> AT -> [*] AT bt command support.

Set the UART port to align with ESP-WROOM-32
Component config -> EPS32-Specific-> [] UART for console output -> Custom.
Component config -> EPS32-Specific-> [] UART TX 17
Component config -> EPS32-Specific-> [*] UART RX 16

Set the correct output (connected with the ESP32 programming port).
Serial Flashed config - /Dev/COM7

Is there something I forget here ?

kurta999, 2020-01-29:
I read somewhere that only wrover supports bt classic due to big memory usage, wrom doesnt have external ram. check my issue what i created a few weeks ago you may find useful answers there.

xcguang, 2020-01-30:
@kurta999 You are right, on the other hand, the firmware will be more larger if enable BT classic. You can compile it as following, which you can find in `.gitlab-ci.yml`, or you enlarge OTA partition size

```
export ESP_AT_MODULE_NAME=WROVER-32
export ESP_AT_PROJECT_PATH=$(pwd)
./esp-idf/tools/idf.py build -DSILENCE=1
```

xcguang, 2020-03-12:
Is the issue solved? May I close it?

closed 2020-03-15 by xcguang