espressif/esp-at #331: Update version in Menuconfig is not in programmed build

Emiel84, 2020-01-27:
When I use Branch v2.0.0.0_esp32 as a baseline and : 
1.Update the AT version in the menuconfig from 2.0.0.0 to 2.0.0.1 (to show a different version).
2. Enable AT BT classic in make menuconfig. 
3. Create a new build 
4. Upload the new build into the ESP-WROOM-32 
Then it is still mentioning the old 2.0.0.0 AT version and build date when requesting the version using the AT version request command.

-----------------------------
AT+GMR<CR><LF>
AT version:2.0.0.0(e44648c - Nov  4 2019 02:18:57)<CR><LF>
SDK version:v3.3-dirty<CR><LF>
compile time(7cac7ad):Jan 27 2020 12:29:14<CR><LF>
Bin version:2.0.1(WROOM-32)<CR><LF>
------------------------

I have programmed the ESP-WROOM-32 using 2 options both having the same result:
1. make flash SILENCE=1  (did align to use COM7 in the menuconfig at step 2 above)
2 make SILENCE=1 + ESP32 DOWNLOAD TOOL V3.6.8

xcguang, 2020-01-30:
Hi, 
You just can change `Bin version`:
### Bin version:2.0.1(WROOM-32)

xcguang, 2020-02-07:
I will close this issue, if you still have question, please reopen it.

closed 2020-02-07 by xcguang