espressif/esp-at #334: [BLE] Unable to connect to server

ikalogic, 2020-01-30:
Hi,

I'm using the version:

```
AT+GMR
AT version:2.0.0.0(e44648c - Nov  4 2019 02:18:57)
SDK version:v3.3
compile time(7cac7ad):Nov  4 2019 06:11:40
Bin version:2.0.0(WROOM-32)
```

When I try to connect to a device having a public MAC address, it works OK.

When I try to connect to a device having a random address, I never works, I trie:

```
AT+BLECONN=0,"ec:b5:6d:78:9a:61",1,10
ERROR (after ~15s)
```
```
AT+BLECONN=0,"ec:b5:6d:78:9a:61",1
ERROR (after ~15s)
```

Here is the SCAN response of that BLE server:

`+BLESCAN:ec:b5:6d:78:9a:61,-63,0201060a095350444e6e41426d6d051206000800020afe,,1`

And by the way here is the response of the device *i am able* to connect to

 `+BLESCAN:88:6b:0f:ce:72:49,-84,020106,,0`

Any idea?

xcguang, 2020-01-30:
Hi,
Sorry for that now we are on the Spring Festival vacation, maybe there is some guys can reply after Feb 13

ikalogic, 2020-01-30:
Thx. Since there are similaritites with #253, i tried loading exactly the same version of the firmware used in that issue (`AT version:1.2.0.0(6fe4345 - Jun 27 2019 05:39:23)`) but i have EXACTLY the same results.

ikalogic, 2020-01-31:
looking at the debug output from the ESP32, i get this:

`E (45977) BLE_AT_GATTC: error:conn_list is empty, delete_node_from_conn_list#L1274`

I'll continue to share my observations here as I try to dig my way through this.

ikalogic, 2020-02-01:
Hi,

I have recompiled esp-at using nimBLE instead of bluedroid stack, and I have implemented my own BLE AT commands. This way it's working, so it's definately a problem in bluedroid stack implementation.

xiewenxiang, 2020-03-12:
Hi,
I'm very sorry for not replying in time.
What is the device for sending broadcast packets? I want to reproduce this issue.

closed 2020-08-10 by xiewenxiang